arduino/arduino-ide #1357: Uploading the sketch OTA unfortunately hangs

kittaakos, 2022-08-26:
### Describe the problem

Originally reported by @MarkSDS [here](https://github.com/arduino/arduino-ide/issues/1331#issuecomment-1228372517):

> Uploading the sketch OTA unfortunately hangs with these messages:
> 
> `Sketch uses 792510 bytes (60%) of program storage space. Maximum is 1310720 bytes. Global variables use 42556 bytes (12%) of dynamic memory, leaving 285124 bytes for local variables. Maximum is 327680 bytes. Sending invitation to 192.168.2.26 Uploading.............` followed by A LOT of dots thereafter. No clue as for the reason for this hanging.

### To reproduce

@kittaakos will update this

### Expected behavior

Can upload the OTA sketch.

### Arduino IDE version

Version: 2.0.0-rc9.2.nightly-20220826 Date: 2022-08-26T03:53:51.674Z CLI Version: git-snapshot [42cd9e17]  

### Operating system

N/A

### Operating system version

N/A

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-08-29:
@MarkSDS which version of the ESP32 boards platform are you using?

You can see it in the Arduino IDE **Boards Manager**, like this:

[image]

MarkSDS, 2022-08-29:
I checked, and I was on 1.0.6:
[image]

Updated towards 2.0.4 (!), thank you.
[image]

Problem appereance altered somewhat, but the result now (one try done) is `No Response`: 
[image]

I came across [this post](https://github.com/espressif/arduino-esp32/issues/1253#issuecomment-433580457) because earlier tonight the module wasn't responsive while on USB neither. That part was solved by unattaching it from the breadboard-to-be. 

I'll investigate further tomorrow and post my findings here. 
Thank you for any guidance.

MarkSDS, 2022-08-29:
Update:
I tried to remove any `Serial.print()` in the OTA sketch. I can imagine it might be hard to write to `Serial`over the air. 
I added some `sys.stderr.write()` in `espota.py` in an attempt to see what's going on, but the Arduino IDE does not use that saved python script while uploading, obviously: the terminal doesn't change it's behaviour. 
Randomly pushing or holding `RST` or `PWR` had no succes.

MarkSDS, 2022-09-01:
Tried the online IDE. 
The connection via WiFi is available, though not stable at all. I can select various boards, e.g. the DOIT ESP32 Devkit V1 as well as the auto-detected `ESP32-c4dd57767370`.

Whenever OTA upload starts, board is recognised as a Yún and compiling `#include <ESPmDNS.h>` throws a fatal error.