arduino/arduino-ide #1359: Cannot upload to board over the air, if wrong password was provided one.

kittaakos, 2022-08-26:
### Describe the problem

When uploading to a MKR1000 based on [this](https://docs.arduino.cc/tutorials/mkr-1000-wifi/wifi-101-ota) tutorial, users have to provide the password. If the password was incorrectly provided the upload with fail with `Failed uploading: uploading error: exit status 1`. Trying to upload again will always fail because the user fields are cached by the IDE2.

### To reproduce

 - Follow [this](https://docs.arduino.cc/tutorials/mkr-1000-wifi/wifi-101-ota) tutorial to upload over the air,
 - When prompted, provide an incorrect password,
 - Upload will fail (expected),
 - Upload again; it will always fail because the fields are cached.

### Expected behavior

Perhaps there should be a meaningful error code/message from the CLI; authentication failed, and then the IDE2 will know that the user fields should be discarded.

### Arduino IDE version

7c361cf

### Operating system

macOS

### Operating system version

12.5.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

ubidefeo, 2022-09-02:
@kittaakos 
when you need to change the password you use the menu item Sketch > Configure and Upload

per1234, 2022-09-02:
Closing because this is the intended behavior: https://github.com/arduino/arduino-ide/pull/550

I have submitted a proposal for how to make the IDE's behavior more intuitive in this case: https://github.com/arduino/arduino-ide/issues/1386

We also need to document this new **Sketch > Configure and Upload** interface, but that documentation is hosted in other repositories, so the tasks are not tracked in this issue tracker.

closed 2022-09-02 by per1234