arduino/arduino-ide #1360: Serial Plotter requires \r\n line endings

sellensr, 2022-08-26:
### Describe the problem

Plotter appears to ignore lines that don't end with both \r and \n in that order. Previous versions worked with lines ending in \n, as does the serial monitor.

I'm not sure if this is an intended behaviour. It isn't mentioned in the tutorial at https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-serial-plotter

### To reproduce

Serial.printf("A:%d,Var2:%d\n",random_variable,static_variable);

and look for output on Serial Plotter. Then change to 

Serial.printf("A:%d,Var2:%d\r\n",random_variable,static_variable);

### Expected behavior

I would expect the plotter to function with any permutation of line endings, e.g. detect \r or \n as an ending, then discard any additional \r or \n until new line begins.

### Arduino IDE version

IDE 2.0 rc9.2

### Operating system

macOS

### Operating system version

12.5.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-08-26:
Thanks for your report @sellensr. This bug is being tracked in the repository that hosts the Serial Plotter codebase: https://github.com/arduino/arduino-serial-plotter-webapp/issues/8

Even though this is a duplicate of that one, I will leave it open for now because I think it will serve as a more effective way of documenting this defect, since users are less likely to be aware of the existence of the `arduino/arduino-serial-plotter-webapp` repository.

nmzaheer, 2022-09-05:
@per1234 Since [arduino/arduino-serial-plotter-webapp#8] (https://github.com/arduino/arduino-serial-plotter-webapp/issues/8) is closed now, can we consider this as closed or is there some testing that needs to be done?

Further, the OP has written that the Serial Plotter should handle all kinds of linebreaks, whereas the Serial Plotter protocol recognises LF "\n" as the only allowed linebreak character.

Should this be captured somewhere in the documentation for users? It is unlikely that users will look at the protocol for their answer.

per1234, 2022-09-05:
>can we consider this as closed

Even though it is resolved, the reason I left this issue open despite there being another in the plotter's repository was with the idea of documenting the bug for Arduino IDE users.

Those users will still be affected until there is a release of the `arduino-serial-plotter-webapp` package and this project's dependency bumped to the version with the fix.

>testing that needs to be done?

Testing is always welcome. I did some today, but it is possible others might catch something I missed.

>OP has written that the Serial Plotter should handle all kinds of linebreaks, whereas the Serial Plotter protocol recognises LF "\n" as the only allowed linebreak character.

Even though not mentioned explicitly in the current documentation, `\r\n` is also supported in Serial Plotter due to this:

https://github.com/arduino/arduino-serial-plotter-webapp/blob/eac6d39f3f5156390bc4f9e09cafcccc121248c9/src/msgAggregatorWorker.ts#L97

https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/parseFloat#description

>If `parseFloat` encounters a character other than a plus sign (`+`), minus sign (`-` U+002D HYPHEN-MINUS), numeral (`0`–`9`), decimal point (`.`), or exponent (`e` or `E`), it returns the value up to that character, ignoring the invalid character and characters following it.

This is also the case in the Arduino IDE 1.x Serial Plotter, and in fact the example code in the ["**SerialPlotter protocol**" document](https://github.com/arduino/Arduino/blob/master/build/shared/ArduinoSerialPlotterProtocol.md) relies on that support ([`Serial.println()` uses `\r\n` line ending](https://github.com/arduino/ArduinoCore-API/blob/1.3.1/api/Print.cpp#L151))

So the only line ending not supported is `\r` alone, as is the case in Arduino IDE 1.x Serial Plotter.

>Should this be captured somewhere in the documentation for users? It is unlikely that users will look at the protocol for their answer.

Excellent point! This important information is not at all easy for the users to find.

I think the correct place to document it here:

https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-serial-plotter

The source of that content is hosted here:

https://github.com/arduino/docs-content/blob/main/content/software/ide-v2/tutorials/ide-v2-serial-plotter/ide-v2-serial-plotter.md

You are welcome to submit an issue to the `arduino/docs-content` repository requesting the documentation, or better yet to submit a pull request contributing the changes to the file directly.

closed 2022-10-04 by per1234