arduino/arduino-ide #1362: MCUFriend Library doesn't compile properly

BSG-SalvoL, 2022-08-27:
### Describe the problem

Compiling the same code using this library with old arduino ide works perfectly, but doing this with new IDE a lot of errors appears considering all variables in macro expansion like undefined. 
Example: ```
c:\Users\xxx\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1210:38: error: 'WR_PIN' was not declared in this scope
 1210 | #define WR_IDLE    PIN_HIGH(WR_PORT, WR_PIN)
```

### To reproduce

Just compile a code with this library

### Expected behavior

The compilation to being successful.

### Arduino IDE version

Arduino IDE 2.0.0-rc9.2

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

ubidefeo, 2022-08-28:
hi @BSG-SalvoL 

Please provide the following:
- link to library and version number
- source code which does not compile on your setup

per1234, 2022-10-11:
Hi @BSG-SalvoL. We are still waiting for you to provide the information requested in the previous reply so that we can proceed with the investigation or evaluation of this.

per1234, 2022-11-26:
Closing due to lack of response.

@BSG-SalvoL if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2022-11-26 by per1234

iasonov, 2023-11-11:
Trying to setup ST7793 400px x 240 px. 
I got errors when I tried to compile example diagnose_TFT_support.ino:

> MCUFRIEND_kbv/utility/mcufriend_shield.h:1216:44: error: 'PIN_HIGH' was not declared in this scope
>  #define CS_IDLE    PIN_HIGH(CS_PORT, CS_PIN)

 
Whole error message:

> 
> In file included from /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:38:0:
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1201:2: error: #error MCU unsupported
>  #error MCU unsupported
>   ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp: In function 'void write24(uint16_t)':
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:87:13: error: 'write8' was not declared in this scope
>      write8(r);
>              ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp: In member function 'void MCUFRIEND_kbv::reset()':
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:95:17: error: 'setWriteDir' was not declared in this scope
>      setWriteDir();
>                  ^
> In file included from /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:38:0:
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1208:31: error: 'RD_PORT' was not declared in this scope
>  #define RD_OUTPUT  PIN_OUTPUT(RD_PORT, RD_PIN)
>                                ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1229:37: note: in expansion of macro 'RD_OUTPUT'
>  #define CTL_INIT()   { GPIO_INIT(); RD_OUTPUT; WR_OUTPUT; CD_OUTPUT; CS_OUTPUT; RESET_OUTPUT; }
>                                      ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:96:5: note: in expansion of macro 'CTL_INIT'
>      CTL_INIT();
>      ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1208:40: error: 'RD_PIN' was not declared in this scope
>  #define RD_OUTPUT  PIN_OUTPUT(RD_PORT, RD_PIN)
>                                         ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1229:37: note: in expansion of macro 'RD_OUTPUT'
>  #define CTL_INIT()   { GPIO_INIT(); RD_OUTPUT; WR_OUTPUT; CD_OUTPUT; CS_OUTPUT; RESET_OUTPUT; }
>                                      ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:96:5: note: in expansion of macro 'CTL_INIT'
>      CTL_INIT();
>      ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1208:46: error: 'PIN_OUTPUT' was not declared in this scope
>  #define RD_OUTPUT  PIN_OUTPUT(RD_PORT, RD_PIN)
>                                               ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1229:37: note: in expansion of macro 'RD_OUTPUT'
>  #define CTL_INIT()   { GPIO_INIT(); RD_OUTPUT; WR_OUTPUT; CD_OUTPUT; CS_OUTPUT; RESET_OUTPUT; }
>                                      ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:96:5: note: in expansion of macro 'CTL_INIT'
>      CTL_INIT();
>      ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1211:31: error: 'WR_PORT' was not declared in this scope
>  #define WR_OUTPUT  PIN_OUTPUT(WR_PORT, WR_PIN)
>                                ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1229:48: note: in expansion of macro 'WR_OUTPUT'
>  #define CTL_INIT()   { GPIO_INIT(); RD_OUTPUT; WR_OUTPUT; CD_OUTPUT; CS_OUTPUT; RESET_OUTPUT; }
>                                                 ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:96:5: note: in expansion of macro 'CTL_INIT'
>      CTL_INIT();
>      ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1211:40: error: 'WR_PIN' was not declared in this scope
>  #define WR_OUTPUT  PIN_OUTPUT(WR_PORT, WR_PIN)
>                                         ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1229:48: note: in expansion of macro 'WR_OUTPUT'
>  #define CTL_INIT()   { GPIO_INIT(); RD_OUTPUT; WR_OUTPUT; CD_OUTPUT; CS_OUTPUT; RESET_OUTPUT; }
>                                                 ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:96:5: note: in expansion of macro 'CTL_INIT'
>      CTL_INIT();
>      ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1214:31: error: 'CD_PORT' was not declared in this scope
>  #define CD_OUTPUT  PIN_OUTPUT(CD_PORT, CD_PIN)
[... 719 lines cut ...]
>          static const uint8_t ILI9341_regValues_ada[] PROGMEM = {        // Adafruit_TFTLCD only works with EXTC=0
>                               ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:2737:30: warning: unused variable 'ILI9481_CPT29_regValues' [-Wunused-variable]
>          static const uint8_t ILI9481_CPT29_regValues[] PROGMEM = {    // 320x430
>                               ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:2749:30: warning: unused variable 'ILI9481_PVI35_regValues' [-Wunused-variable]
>          static const uint8_t ILI9481_PVI35_regValues[] PROGMEM = {    // 320x480
>                               ^
[... cut ...]
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:2786:30: warning: unused variable 'ILI9481_RGB_regValues' [-Wunused-variable]
>          static const uint8_t ILI9481_RGB_regValues[] PROGMEM = {    // 320x480
>                               ^
> /Users/iasonov/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:2878:7: warning: label 'common_9488' defined but not used [-Wunused-label]
>        common_9488:
>        ^
> 
> exit status 1
> 
> Compilation error: exit status 1
>