arduino/arduino-ide #1363: not recoverable PROXY Settings

pippo-75, 2022-08-28:
### Describe the problem

Click on File -> Preferences -> Network tab: then set to -Manual proxy configuration --> SOCKS- and check OK. 
Next, close the IDE.
Reopen the IDE and check in File -> Preferences -> Network tab.
If your IDE works like mine, it will no longer be possible to go back to No Proxy.

### To reproduce

As described above

### Expected behavior

Must accept and hold the -No proxy- setting.

### Arduino IDE version

2.0.0-rc9.2.nightly-20220825

### Operating system

Windows

### Operating system version

Windows 10 pro 64 bit 21H2 19044.1889

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-08-29:
Hi @pippo-75. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-cli/issues/1677.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

---

I will share the workaround for this issue:

1. Select **File > Quit** from the Arduino IDE menus if it is running.
1. Use any text editor to open the file at the following path:
   - **Windows:**
     ```text
     C:\Users\<username>\.arduinoIDE\arduino-cli.yaml
     ```
   - **Linux:**
     ```text
     ~/.arduinoIDE/arduino-cli.yaml
     ```
   - **macOS:**
     ```text
     ~/.arduinoIDE/arduino-cli.yaml
     ```
   ❗ The `.arduinoIDE` folder may be hidden by default by your operating system.
1. Delete the lines from the file that have this format:
   ```yaml
   network:
     proxy: http://pippo:pluto@example.com:1234/
     user_agent_ext: daemon
   ```
1. Save the file.
1. Start the Arduino IDE.

You should now find that "**No proxy**" is selected in the "**Network**" tab of the "**Preferences**" dialog.

closed 2022-08-29 by per1234

pippo-75, 2022-08-29:
Yes that done the job. A big thank you.