arduino/arduino-ide #1394: Non-functional items in editor context menu

per1234, 2022-09-04:
### Describe the problem

Right clicking on the sketch editor view opens a context menu that offers various useful operations that can be done on the selection.

🐛 The "**Refactor...**" and "**Source Action...**" context menu items don't appear to be usable with Arduino sketches. They increase the complexity and clutter of the Arduino IDE user interface without adding anything of value.

### To reproduce

1. Select any code in the sketch editor view.
1. Right click on the selection.
1. Select "**Refactor...**" from the context menu.
   😕 A tooltip appears: "**No refactorings available**"
1. Right click on the selection.
1. Select "**Source Action...**" from the context menu.
   😕 A tooltip appears: "**No source actions available**"

### Expected behavior

All items in the editor context menu are useful.

### Arduino IDE version

2.0.0-rc9.2.snapshot-20cc34c

### Operating system

Windows, Linux

### Operating system version

Windows 10, Ubuntu 20.04

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-09-05:
> 🐛 The "**Refactor...**" and "**Source Action...**" context menu items don't appear to be usable with Arduino sketches.

The LS should use them. Instead of removing the context menu items, I recommend supporting them in IDE2 in the long run.

per1234, 2022-09-05:
The question is: how difficult will it be to support them, and how long will it take to add that? If it will be a long time and the context menu customization is easy to do, I think it would be worth removing them provisionally.

>all over the application

I would remove them anywhere they are not working.

I also don't think it is worth making any effort or using system resources to provide enhanced functionalities for JSON (though I would make an exception for the `settings.json` schema). Only a tiny fraction of users will want to directly edit JSON files in the IDE. If they need enhanced JSON capabilities, they can use any of the many existing tools that provide it.

kittaakos, 2022-09-06:
> I would remove them anywhere they are not working.

IDE2 does not know it upfront. If a language (`json`, `ino`) is contributed to IDE2, the capabilities come from the server the menu items will be there. IDE2 has no chance to figure out if the capabilities are functional or not.

> The question is: how difficult will it be to support them

I think the problem is [here](https://github.com/arduino/arduino-language-server/blob/main/ls/ls.go#L336). The Arduino LS remaps the code actions from clangd as they are, but the Arduino LS's id is not `clangd`, but the language ID is [`ino`](https://github.com/kittaakos/vscode-arduino-tools/blob/2c458f1f77ad95f6278136bf0383d2c2c504ffd4/src/extension.ts#L266) so the quick fixes, refactorings won't work.

> If it will be a long time and the context menu customization is easy to do,

Customizing these menus is not trivial either. I propose sticking to the natural workflow and fixing the real issue instead of trying to remove the menu items as part of a hack.

per1234, 2022-09-06:
@ubidefeo, since this was originally your issue (9890 in the internal backlog), what do you think? Should we close this in favor of a bug report in `arduino/arduino-language-reference` tracking the defect that causes these context menu items to be non-functional?

berparis, 2022-12-12:
Hello,
I don't understand what exact status now is that bug report, but it would be very usefull to have this feature working.
I'm using Arduino IDE 2.0.3 with MacOS Ventura 13.0.1 and I observe exactly the described problem here on top of page.
Thanks, Bernard

commit 192aac5a (2023-05-09): chore: updated to Theia `1.37.0`

 - Updated `@theia/*` to `1.37.0`.
 - Fixed all `yarn audit` security vulnerabilities.
 - Updated to `electron@23.2.4`:
   - `contextIsolation` is `true`,
   - `nodeIntegration` is `false`, and the
   - `webpack` target is moved from `electron-renderer` to `web`.
 - Updated to `typescript@4.9.3`.
 - Updated the `eslint` plugins.
 - Added the new `Light High Contrast` theme to the IDE2.
 - High contrast themes use Theia APIs for style adjustments.
 - Support for ESM modules: `"moduleResolution": "node16"`.
 - Node.js >= 16.14 is required.
 - VISX langage packs  [... cut]