arduino/arduino-ide #1395: Remove "Format Selection" from editor context menu

per1234, 2022-09-04:
### Describe the request

Remove the "**Format Selection**" item from the editor context menu.

🙂 The IDE's user interface will be made more streamlined without significant loss of functionality. The capability will still be available via the keyboard shortcut and "**Format Selection**" command palette item if it is wanted by some users.

### Describe the current behavior

Right clicking on the sketch editor view opens a context menu that offers various useful operations that can be done on the selection:

[image]

There are two items in the menu for formatting code:

- **Format Document**
- **Format Selection**

The first provides the essential functionality of formatting the entire sketch file, more commonly accessed via the **Tools > Auto Format** menu item or <kbd>**Ctrl**</kbd>+<kbd>**T**</kbd> keyboard shortcut.

The second provides the more niche capability of applying formatting only to the text currently selected in the editor view. It is not clear that this is something of use to any significant number of users.

### Arduino IDE version

2.0.0-rc9.2.snapshot-20cc34c

### Operating system

Windows, Linux

### Operating system version

Windows 10, Ubuntu 20.04

### Additional context

Related: https://github.com/arduino/arduino-ide/issues/1394

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

kittaakos, 2022-09-05:
Is this required for `.ino` files only or remove all the IDE2? For example, open the `settings.json` file and check the context menu.

per1234, 2022-09-05:
The GitHub issue was created by me, but I actually only published it from the internal backlog (9890, I split it into two separate issues since there appeared to be distinct motivations for the requests: not working vs. not needed).

I see @ubidefeo was the original creator of the issue, so maybe they can answer the question.

per1234, 2022-11-22:
Since we never received any feedback after 2.5 months, I'm going to close this as declined. The feature works fine.

I don't think the feature is at all useful to the average Arduino IDE user, but there is a use case for this feature in specific circumstances: Sometimes you are contributing to a codebase with inconsistent formatting due to sloppy prior development work. As a conscientious developer, you want to correctly format the code you are contributing. However, it would not be appropriate to submit formatting changes to unrelated code with your contribution. In this case, doing a universal format and then having to sort your contribution out of the resulting huge diff when staging the commit can be quite some work. Being able to selectively format only your own code allows you to avoid that.

closed 2022-11-22 by per1234