arduino/arduino-ide #1396: Please make port selections little bit stickier

Master811129, 2022-09-04:
### Describe the problem

Every time i want to upload my sketch I must select the correct port. it is because Arduino leonardo disconnects from usb and immediately falls back to wrong port.

### To reproduce

- attach an Arduino leonardo on linux
- select the correct port
- after uploading the sketch port selector falls back to wrong port

### Expected behavior

Port selector should wait at least 2 seconds for the Arduino Leonardo to re-appear and then start the fallback process.

### Arduino IDE version

Version: 2.0.0-rc9.2 Date: 2022-08-10T13:01:55.702Z CLI Version: 0.26.0-rc.1 [fc2ea723] 

### Operating system

Linux

### Operating system version

Rolling BTW.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-04:
Hi @Master811129. Thanks for your report

>Version: 2.0.0-rc9.2 Date: 2022-08-10T13:01:55.702Z CLI Version: 0.26.0-rc.1 [fc2ea723]

This is a significantly outdated version of the Arduino IDE. Most importantly, it is missing the related fixes made in this PR: https://github.com/arduino/arduino-ide/pull/1328

Please try it again with the latest nightly build and then comment here to let us know whether or not the bug still occurs.

The nightly build download links are listed here:
https://www.arduino.cc/en/software#nightly-builds

Master811129, 2022-09-05:
Yes, the latest AppImage doesn't have this issue. sorry for interrupting.

closed 2022-09-05 by Master811129