arduino/arduino-ide #1403: Library/Boards Manager "Install" button sometimes partial or missing

per1234, 2022-09-06:
### Describe the problem

When the mouse pointer is hovered over one of the Library/Boards Manager or list items, a version menu and "**INSTALL**" button are added at the bottom of the item, which increases its vertical size.

🐛 When the length of the listings fits within the vertical dimension of the view when not hovered, but does not fit when hovered, some glitches can occur.

### To reproduce

1. Open the "**Library Manager**" view.
1. Resize the horizontal width of the view to the minimum.
   **ⓘ** This is the width set by the IDE by default.
1. Type `arduinoiotcloud` in the "**Filter your search...**" field.
   **ⓘ** This is only an arbitrary example of a search that produces a suitable length of listings. By chance it has the perfect length for the default window and view dimensions on my machine.
1. Resize the vertical dimension of the Arduino IDE window so that it is just long enough to show the full listings while in the non-hovered state, without a scrollbar on the Library Manager view.
   [image]
1. ❗ Without passing the mouse pointer over any of the other library listings in the view along the way ❗, move the mouse pointer from an area of the Arduino IDE window outside the "**Library Manager**" view to hover the mouse pointer over "**Arduino_ConnectionHandler**", .
   🐛 Only half the "**INSTALL**" button is shown:
   [image]
1. ❗ Without passing the mouse pointer over any of the other library listings in the view along the way ❗, move the mouse pointer from an area of the Arduino IDE window outside the "**Library Manager**" view to hover the mouse pointer over "**ArduinoIoTCloud**", .
   🐛 The "**INSTALL**" button is not shown:
   [image]

### Arduino IDE version

2.0.0-rc9.3

### Operating system

Windows 10

### Additional context

The issue was introduced by https://github.com/arduino/arduino-ide/pull/1385 (does not occur when using the build from https://github.com/arduino/arduino-ide/commit/df3a34eec608230b213f0b595078fc496d3e6245)

---

The reason why the issue only occurs with the view at or near the minimum width seems to be related to the fact that, at this width, the position of the "**INSTALL**" button is below the version menu:

[image]

While at any greater width it is to the side of the version menu:

[image]

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-09-06:
> the position of the "**INSTALL**" button is below the version menu:

Correct. On a scrollbar visibility change, the position of the `INSTALL` button changes, and the required height calculation will be incorrect.

per1234, 2022-10-26:
Fixed by https://github.com/arduino/arduino-ide/pull/1568

closed 2022-10-26 by per1234