arduino/arduino-ide #1404: Serial Backlog

philipmcgaw-cpi, 2022-09-06:
### Describe the request

historically, you could 'up arrow' through the recently sent commands, to send the same one a second time, or even go further back in history.

### Describe the current behavior

none, this functionality is missing.

### Arduino IDE version

IDE 2.0.0-rc9.3

### Operating system

macOS

### Operating system version

Monterey

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

philipmcgaw-cpi, 2022-09-06:
This issue is linked to a forum post: https://forum.arduino.cc/t/ide-2-0-0-rc9-3-serial-console-text-entry/1028868/3

nmzaheer, 2022-09-09:
The legacy IDE has a separate class called commandHistory that implements this functionality at [CommandHistory.java](https://github.com/arduino/Arduino/blob/57a931c9c4c084057417837239ad8136f8a7b1aa/app/src/processing/app/CommandHistory.java#L10-L14)
```java
public class CommandHistory {

  private final LinkedList<String> commandHistory = new LinkedList<String>();
  private final int maxHistorySize;
  private ListIterator<String> iterator = null;
```
which is consumed by [SerialMonitor.java](https://github.com/arduino/Arduino/blob/57a931c9c4c084057417837239ad8136f8a7b1aa/app/src/processing/app/SerialMonitor.java#L76-L101) as

```java
      public void keyPressed(KeyEvent e) {
        switch (e.getKeyCode()) {

          // Select previous command.
          case KeyEvent.VK_UP:
            if (commandHistory.hasPreviousCommand()) {
              textField.setText(
                  commandHistory.getPreviousCommand(textField.getText()));
            }
            break;

          // Select next command.
          case KeyEvent.VK_DOWN:
            if (commandHistory.hasNextCommand()) {
              textField.setText(commandHistory.getNextCommand());
            }
            break;

          // Reset history location, restoring the last unexecuted command.
          case KeyEvent.VK_ESCAPE:
            textField.setText(commandHistory.resetHistoryLocation());
            break;
        }
      }
    });
```
----
### Proposal
1. I'm not sure whether Javascript has a built-in standard LinkedList implementation which means we will have to write an implementation of a LinkedList and then incorporate it into the Serial monitor.
2. We can implement the same functionality with an Array, however if you look at the commit history of commandHistory.java the codebase has moved from an initial Array implementation to the LinkedList version for performance reasons.

Looking forward to comments on how to proceed with this.

PR #1562 (2022-10-25): #1404 Added Message History to Serial Monitor
### Motivation
Feature added for message history in Serial Monitor

### Change description
Utilizes the RingList implementation of @dwightfowler-cd  [here](https://github.com/dwightfowler-cd/arduino-ide/blob/Issue-1404/arduino-ide-extension/src/browser/serial/monitor/serial-monitor-send-input.tsx) with some corrections for traversal of the list

### Other information
- Another PR for this is available at PR #1467
- This closes #1404 
### Reviewer checklist

* [x] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [... cut]
closed 2022-10-25 by kittaakos