arduino/arduino-ide #1405: Character encoding problem with Serial Monitor

nbourre, 2022-09-06:
### Describe the problem

When using serial printing accented characters or ASCII higher than 127, the serial monitor mostly shows a "�", but not always.

[image]

### To reproduce

`Serial.print` accented characters or ASCII with values higher than 127.

E.g.:
```cpp
    Serial.print("Humidité: ");
    Serial.print(h);
    Serial.print(("%  Température: "));
    Serial.print(t);
    Serial.print(("° C "));
    Serial.print(("Humidex : "));
    Serial.print(humidex);
    Serial.println(("° C "));
```

### Expected behavior

The serial monitor should show the characters in the right encoding or at least UTF-8.

### Arduino IDE version

2.0.0-rc9.3

### Operating system

Windows

### Operating system version

11

### Additional context

Full code example :
```cpp
#include <DHT.h>

// Broche de données
#define DHTPIN 2

// Type de capteur pour la librairie
#define DHTTYPE DHT11  // DHT 11

// Déclaration de l'objet
DHT dht(DHTPIN, DHTTYPE);

long currentTime = 0;
long dhtPrevious = 0;
long dhtDelay = 1000;

void setup() {
  Serial.begin(9600);
  Serial.println(F("DHTxx test!"));

  dht.begin();  // Initialisation

  currentTime = millis();
}

void loop() {
  currentTime = millis();

  if (currentTime - dhtPrevious >= dhtDelay) {
    dhtPrevious = currentTime;

    float h = dht.readHumidity();
    float t = dht.readTemperature();
    float humidex = dht.computeHeatIndex(t, h, false);

    Serial.print("Humidité: ");
    Serial.print(h);
    Serial.print(("%  Température: "));
    Serial.print(t);
    Serial.print(("° C "));
    Serial.print(("Humidex : "));
    Serial.print(humidex);
    Serial.println(("° C "));
  }
}
```

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-06:
Hi @nbourre. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/589.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-09-06 by per1234

Serhioromano, 2022-12-26:
I have the same issue.

```cpp
SoftwareSerial DebugSerial(8,9);

// setup
DebugSerial.begin(9600);

// loop
DebugSerial.println("Light: ");
```
And here is what I get

```
[Starting] Opening the serial port - COM11
U�U�U�U�U�U�U�U�U�U�U�U�U�U�U�U�U�U�U�U�U�U�U�U�
```