arduino/arduino-ide #1416: IDE2.0.0-RC9.4 Excessive memory occupation

13507086298, 2022-09-09:
### Describe the problem

After editing this version for a period of time, I will occupy all the memory of the computer, and the CPU will be high.
[image]
[image]
[image]
[image]

### To reproduce

After I open the IDE to compile, the memory of memory will be soaring.

### Expected behavior

I have no idea

### Arduino IDE version

2.0.0-RC9.4

### Operating system

Windows

### Operating system version

windows 10 21H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-13:
Hi @13507086298. Thanks for your report.

Which board (or boards) do you have selected in the Arduino IDE when this issue occurs?

Does the issue occur when you are editing a specific sketch (or sketches), or does it occur even with the basic sketch you get from **File > New**?

>After editing this version for a period of time

What is the approximate period of time?

13507086298, 2022-10-01:
Sorry, I haven't logged in to github recently because of drawing pcb boards.
The development board I chose was uno, because I was using 1.8.19 and 2.0rc9.4 at the same time. At a certain moment, I suddenly found that the memory usage was abnormal. After I closed the two IDEs and restarted the operating system, many times Only start 2.0rc9.4 alone, and the abnormal memory usage occurs immediately.
I didn't try to create a new sketch after this.