arduino/arduino-ide #1420: RC9.4 Serial Port Resets Board

sayzergithub, 2022-09-11:
### Describe the problem

Once the serial monitor is opened, and then closed later on, port stays open while IDE is running.
And then, when the IDE is closed, it sends a reset pulse to the board via serial port.

### To reproduce

Connect a board.
Open the serial monitor, and then close it.
Now close IDE.
Board resets.

### Expected behavior

reset to the board.

### Arduino IDE version

rc9.4

### Operating system

Windows

### Operating system version

Win10

### Additional context

Where can I submit a suggestion for IDE?

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-13:
Hi @sayzergithub. Thanks for your report.

I did some experimentation but was not able to reproduce the issue.

>port stays open while IDE is running

What are you doing to determine that the port stays open?

>Connect a board.

Which specific board (or boards) have you encountered this issue with?

>Board resets.

What are you doing to determine that the board reset?

sayzergithub, 2022-09-13:
"What are you doing to determine that the port stays open?"
Using an external serial port monitor, it can not access the port.

"Which specific board (or boards) have you encountered this issue with?"
ESP32 with SILAB CP2102 USB-Serial port chip.

"What are you doing to determine that the board reset?"
Immediately after IDE is closed, ESP32 board restarts.

per1234, 2022-09-14:
Thanks for the clarifications. In order to progress with the investigation, I'm going to ask you to try an experiment and then report your results here.

Please try this:

1. Open the Windows "**Task Manager**":
   https://techcommunity.microsoft.com/t5/windows-11/how-to-run-task-manager-on-windows-11-6-ways/m-p/2701239
1. If it is not already in that mode, click the "**More details**" button at the bottom of the **Task Manager** window.
1. Select the "**Details**" tab in the **Task Manager** window.
1. Start the Arduino IDE.
1. Select the board and port for your ESP32.
1. Open **Serial Monitor**.
1. Switch to the **Task Manager** window.
1. Verify that you can see a process with the name "**serial-monitor.exe**".
1. Switch to the Arduino IDE window.
1. Close **Serial Monitor** in the Arduino IDE window.
1. Switch to the **Task Manager** window.

After following these instructions, do you still see a process with the name "**serial-monitor.exe**" in the Windows "**Task Manager**"?

per1234, 2022-10-11:
Hi @sayzergithub. Did you ever get around to trying the experiment I requested in my previous reply? If so, what was the result?

sayzergithub, 2022-10-11:
Hello,
I am having some health issues with my kidneys.
I will try your suggestion as soon as I am up and running again.:)

Thank you,

Ömer YILDIZ

MCU Embedded Systems Designer

IT Professional
+90 539 822 9258
linkedin<https://www.linkedin.com/in/omeryildizlinkedin/>

Doğanet Elektronik ve Bilişim Sis. San. Tic. Ltd. Şti.
Perpa Tic. Merk B Blok Kat 11 No 1589 Şişli-Istanbul
Tel&Fax: 0212 210 1160

per1234, 2023-09-28:
Since we didn't receive any additional reports or the additional information needed to investigate, I'll close this issue. If anyone is experiencing this problem with the latest version of Arduino IDE, please comment here and we will continue with the investigation.

---

Note there is a confirmed bug that causes the equivalent problem with Serial Plotter. That is tracked in the dedicated issue https://github.com/arduino/arduino-ide/issues/1423

closed 2023-09-28 by per1234