arduino/arduino-ide #1429: Suppress error notifications for failures of convenience operations resulting from lack of Internet access

per1234, 2022-09-13:
### Describe the request

Do not show error notifications for failures of automated non-essential operations due to lack of Internet access.

Lack of Internet access should only be considered an error condition when it causes the failure of an operation that falls under one of the following categories:

- Essential to core functionality of Arduino IDE (e.g., download of **serial-discovery** tool on first run)
- User triggered action (e.g., installing a library via Library Manager)

🙂 User will not be bothered by the appearance of unnecessary error notifications.

### Describe the current behavior

Users won't necessarily have Internet access on every single startup of the IDE. Although the IDE's functionality is enhanced by the ability to access resources from the Internet, the core functionality of sketch development does not have a reliance on an Internet connection. So Internet access is not considered a prerequisite for ongoing usage of the IDE.

On every startup, the IDE reads some information from the Internet. On the first run, this includes some essential files, and communicating to the user when these could not be downloaded is important (https://github.com/arduino/arduino-ide/issues/784). However, on all other runs the information accessed is only for non-essential features:

- Update the indexes used by Boards Manager and Library Manager.
- Check if there is a newer version of the Arduino IDE available.

Multiple prominent error notifications are shown on every startup when these operations fail due to lack of Internet access.

[image]

🙁 The cryptic index download error might give users might the impression that something is wrong with the IDE.

🙁 Users might be given the impression that the IDE can only be used while connected to the Internet.

### Arduino IDE version

2.0.0-rc9.4

### Operating system

Windows

### Operating system version

10

### Additional context

For the sake of searchability, I'll add the text of the notifications:

```text
Error downloading index 'https://downloads.arduino.cc/packages/package_index.tar.bz2': Get "https://downloads.arduino.cc/packages/package_index.tar.bz2": dial tcp: lookup downloads.arduino.cc: no such host
```

```text
Error while checking for Arduino IDE updates. Request checkForUpdates failed with message: net::ERR_NAME_NOT_RESOLVED
```

```text
Error while checking for Arduino IDE updates. Request checkForUpdates failed with message: net::ERR_INTERNET_DISCONNECTED
```

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

Franck78, 2022-09-14:
if we are speaking of this notification windows, please add an option to never show it again.

It is so useless and annoying because it hides the output, makes a distraction for your eyes and 99%, the error is dumb and don't need copy.
Select text from the output windows is enough !!!
[image]

kittaakos, 2022-09-29:
Easier if IDE2 runs the update checks only if [`navigator.onLine`](https://developer.mozilla.org/en-US/docs/Web/API/Navigator/onLine). Otherwise, it doesn't. No error handling would be required.

IDE2 already warns the users if there is no Internet connection.

[image]

https://github.com/arduino/arduino-ide/blob/6f077173693cdfd1ed2a65a5754f05b4fcca8ac6/arduino-ide-extension/src/browser/arduino-frontend-contribution.tsx#L69

https://github.com/arduino/arduino-ide/blob/6f077173693cdfd1ed2a65a5754f05b4fcca8ac6/arduino-ide-extension/src/browser/widgets/cloud-sketchbook/cloud-user-status.tsx#L120

AlbyIanna, 2022-10-04:
@kittaakos unfortunately using `navigator.onLine` is not a safe indicator as it can result in [false positives](https://www.electronjs.org/docs/latest/tutorial/online-offline-events): 

> Since many cases return true, you should treat with care situations of getting false positives, as we cannot always assume that true value means that Electron can access the Internet. For example, in cases when the computer is running a virtualization software that has virtual Ethernet adapters in "always connected" state. Therefore, if you want to determine the Internet access status of Electron, you should develop additional means for this check.

Then I believe we should handle those errors somehow. It would also be nice to change those [two](https://github.com/arduino/arduino-ide/blob/6f077173693cdfd1ed2a65a5754f05b4fcca8ac6/arduino-ide-extension/src/browser/arduino-frontend-contribution.tsx#L69) [lines](https://github.com/arduino/arduino-ide/blob/6f077173693cdfd1ed2a65a5754f05b4fcca8ac6/arduino-ide-extension/src/browser/widgets/cloud-sketchbook/cloud-user-status.tsx#L120) to use a more reliable way of checking the internet connection, but I guess we can handle that separately.

kittaakos, 2022-10-06:
> unfortunately using `navigator.onLine` is not a safe indicator as it can result in [false positives](https://www.electronjs.org/docs/latest/tutorial/online-offline-events):

Thanks for the link. What about trying [`is-reachable`](https://www.npmjs.com/package/is-reachable) or [`is-online`](https://www.npmjs.com/package/is-online)? I am sure a lib with 100k downloads per week will perform better than anything we do in-house.

AlbyIanna, 2022-10-06:
Thank you, I will do some tests with those packages.