arduino/arduino-ide #1431: Add missing translations

kittaakos, 2022-09-14:
### Describe the problem

The following problems were reported by users:
[image]
[image]
[image]

### To reproduce

Open 2.0.0 IDE2, and set the language to non-English.

### Expected behavior

All text defined by IDE2/Theia must be translated when a translation exists.

----

> All text defined by IDE2/Theia

Not from the CLI.

### Arduino IDE version

2.0.0

### Operating system

N/A

### Operating system version

n/a

### Additional context

Go through the code and add the missing translations.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-14:
The equivalent version of [**Eclipse Theia** IDE framework](https://theia-ide.org/) (1.25.0), with the same version of the language pack ([`vscode-language-pack-zh-hans@1.53.2`](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-zh-hans)) has all these strings localized:

[image]

[image]

[image]

kittaakos, 2022-09-14:
`Replace` in the SIW widget:

[image]

kittaakos, 2022-09-14:
Typo in translations keys:
```
      "succesfullyInstalledPlatform": "已成功安装平台 {0}：{1}",
      "succesfullyUninstalledPlatform": "已成功卸载平台 {0}：{1}"
```

`succesfully` -> `successfully`

g-castronovo, 2022-09-15:
The following strings are currently missing in en.json source file, therefore they are missing in all langs. It would be cool if we could add them next time.
[image]

g-castronovo, 2022-09-15:
This string isn't missing 100%. the green button is the same content of the first sentence "**Error while burning the bootloader**". then for some reasons it's switching to Italian - which is my system operator. "missing the programmer"? No idea what the source is .....
[image]

g-castronovo, 2022-09-15:
Menu bar to search inside the sketch- **Find** (translation is done but not present in the IDE) **Replace** it's not inside en.json, (it used to be) **Find in selection** it's not inside en.json; "Close (escape)"  - this is a weird  and not present in the IDE. I swear some of these strings used to be done...
[image]

kittaakos, 2022-09-15:
> then for some reasons it's switching to Italian

> me reasons it's switching to Italian - which is my system operator language. "missing the programmer"? No idea what the source is

The source is the CLI. IDE2 toasts the error message as-is from the CLI. There is a caveat here, IDE2 and CLI language can be configured separately.  🤦 

I expect the following is going on:
 - copy error message [does have a translation](https://github.com/arduino/arduino-ide/blob/5f82577bc19fa38a3f24fa27c1c2f6a01a977083/i18n/tr.json#L154), your IDE2 language is Turkish,
 - the CLI language is Italian (see under `~/arduinoIDE/arduino-cli.yaml`, I bet you have `locale: it` there),
 - and I do not know why it prints `Error while burning bootloader` instead of [`Bootloaderı yüklerken hata:`](https://github.com/arduino/arduino-ide/blob/5f82577bc19fa38a3f24fa27c1c2f6a01a977083/i18n/tr.json#L47)

Thank you for reporting it.

kittaakos, 2022-09-15:
> **Replace** it's not inside en.json, (it used to be) **Find in selection** it's not inside en.json; "Close (escape)" - this is a weird and not present in the IDE. I swear some of these strings used to be done...

It's logged here https://github.com/arduino/arduino-ide/issues/1431#issuecomment-1246905029.

g-castronovo, 2022-09-15:
**All** is translated inside Transifex but not in the IDE
**Default** and **More** are missing strings in the en.json
[image]

per1234, 2022-09-15:
> I swear some of these strings used to be done...

This can be caused by changes to the names of the internationalization keys in the Arduino IDE codebase.

For example, when the typo Akos reported [here](https://github.com/arduino/arduino-ide/issues/1431#issuecomment-1246956375) is fixed in the codebase, the string will go back into an untranslated state on Transifex. Hopefully translation memory can make any re-translations a simple process at least.

g-castronovo, 2022-09-15:
Yes if it is identical, it will automatically filled in! No problem! It's just I cannot find those strings "More" and "Default" aren't' listed anymore inside Transifex. But I could be confused with IDE 2.0 or older IDE1.5 localisation

kittaakos, 2022-09-16:
> The equivalent version of [**Eclipse Theia** IDE framework](https://theia-ide.org/) (1.25.0), with the same version of the language pack ([`vscode-language-pack-zh-hans@1.53.2`](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-zh-hans)) has all these strings localized:

For the record, it does not work for me with Theia 1.25.0 and the Chinese language pack manually installed:

[image]

kittaakos, 2022-09-16:
I have found the cause of why `Edit` is missing. As of writing, IDE2 uses Theia 1.25.0, it requires the following translation entry to be present in the translation file (`./plugins/vscode-language-pack-zh-hans/extension/translations/main.i18n.json`):

```json
{
  "vs/workbench/contrib/notebook/common/model/notebookTextModel": {
    "defaultEditLabel": "编辑"
  },
}
```

This entry 👆 is available in an older version of the Chinese language pack. Go to https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-zh-hans/1.53.2, download the VS Code extension, unpack it, and check the file.

This entry is unavailable in a more recent version of the Chinese language pack. See for example https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-zh-hans/1.71.0.

```json
"vs/workbench/api/browser/mainThreadCustomEditors": {
  "defaultEditLabel": "编辑"
},
```

In summary, IDE2's Theia version (1.25.0) cannot deal with the newer version of language packs.

I see three options:
 - Bump the Theia version: it's a more extensive effort.
 - Downgrade the language pack versions: we need to check if everything is correct manually.

I will leave the decision to @davegarthsimpson.

The rest of the translation issues are fixed.

per1234, 2022-09-16:
>IDE2's Theia version (1.25.0) cannot deal with the newer version of language packs.

Good catch. I somehow missed that the language pack versions had been bumped in the IDE.

>Leave as-is until we make the Theia version bump.

Is that even really a guaranteed solution? Theia 1.29.0 is still using VS Code API 1.53.2:

https://github.com/eclipse-theia/theia/blob/v1.29.0/dev-packages/application-package/src/api.ts#L21

It seems to me that the only correct approach is to keep language packs synced with the VS Code API version in use.

kittaakos, 2022-09-16:
> > Leave as-is until we make the Theia version bump.
> 
> Is that even really a guaranteed solution? Theia 1.29.0 is still using VS Code API 1.53.2:

Good point. We should probably start a discussion on the Theia forum.

> It seems to me that the only correct approach is to keep language packs synced with the VS Code API version in use.

In this case, we have to go back to [this commit](https://github.com/microsoft/vscode-loc/commit/b9f51641f75b7ebcd16e5a8776342d8cacbc2d38). The `vscode` engine version matches what Theia supports.

commit 9ef04bb8 (2022-09-26): Fixed missing translations

Aligned the languge pack versions.

Closes #1431
closed 2022-09-26 by kittaakos