arduino/arduino-ide #1433: Property 'upload.tool.serial' is undefined with SAMD processors

supremeneuron, 2022-09-15:
### Describe the problem

I have a SAMD21 based board that I upload projects to.  I have been using the 1.8.13 board library on a Windows 11 machine just fine with the Arduino 1.8.19 development environment.  Today I installed the new 2.0.0 Arduino IDE.  I've managed to successfully compile the projects, but nothing will upload.  I repeatedly get a ....

**Property 'upload.tool.serial' is undefined**  - Error that I cannot seem to figure out.   I have uninstalled the 2.0.0 IDE and re-installed it to no avail.  I have even installed it on another machine with identical results.  Everything uploads fine with 1.8.19 IDE on the same machine, but the 2.0.0 IDE simply fails with the same error. 

### To reproduce

As outlined above.  Attempt to upload a SAMD21 based project in the 2.0.0 IDE and an upload error results.

### Expected behavior

Should simply upload without issues similar to the 1.8.19 IDE.

### Arduino IDE version

2.0.0

### Operating system

Windows

### Operating system version

11 Pro

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-15:
Hi @supremeneuron. Thanks for your report.

In order to gather more information that might help us to troubleshoot your problem, I'm going to ask you to provide the full output from the upload when in verbose mode.

Please do this:

1. Select **File > Preferences** from the Arduino IDE menus.
1. Check the box next to **Show verbose output during > ☐ compilation**.
1. Check the box next to **Show verbose output during > ☐ upload**.
1. Click the <kbd>**OK**</kbd>** button.
1. Do an upload, just as you did before.
1. Wait for upload to fail.
1. After the upload fails you'll see a button on the "**Upload error**" notification at the bottom right corner of the IDE window: <kbd>**COPY ERROR MESSAGES**</kbd>. Click that button.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here.

supremeneuron, 2022-09-16:
Thank for the prompt response.  Below is the information you requested.

____________________________________________________________________________________________________________________

Using board 'arduino_zero_native' from platform in folder: C:\Users\jpsdr\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13
Using core 'arduino' from platform in folder: C:\Users\jpsdr\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13
Detecting libraries used...
"C:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Zero\"" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\cores\\arduino/api/deprecated" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\cores\\arduino" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\variants\\arduino_zero" "C:\\Users\\jpsdr\\AppData\\Local\\Temp\\arduino-sketch-281CA033AE8E842095AE6225B3564EE8\\sketch\\CapTouchPaint.ino.cpp" -o nul
Alternatives for Adafruit_GFX.h: ***@***.***
ResolveLibrary(Adafruit_GFX.h)
  -> candidates: ***@***.***
"C:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Zero\"" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\cores\\arduino/api/deprecated" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\cores\\arduino" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\variants\\arduino_zero" "-IC:\\Users\\jpsdr\\Documents\\Arduino\\libraries\\Adafruit-GFX-Library-master" "C:\\Users\\jpsdr\\AppData\\Local\\Temp\\arduino-sketch-281CA033AE8E842095AE6225B3564EE8\\sketch\\CapTouchPaint.ino.cpp" -o nul
Alternatives for SPI.h: ***@***.***
ResolveLibrary(SPI.h)
  -> candidates: ***@***.***
"C:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Zero\"" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\cores\\arduino/api/deprecated" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\cores\\arduino" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\variants\\arduino_zero" "-IC:\\Users\\jpsdr\\Documents\\Arduino\\libraries\\Adafruit-GFX-Library-master" "-IC:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\libraries\\SPI" "C:\\Users\\jpsdr\\AppData\\Local\\Temp\\arduino-sketch-281CA033AE8E842095AE6225B3564EE8\\sketch\\CapTouchPaint.ino.cpp" -o nul
Alternatives for Adafruit_ILI9341.h: ***@***.***
ResolveLibrary(Adafruit_ILI9341.h)
  -> candidates: ***@***.***
"C:\\Users\\jpsdr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10 [... cut ...]
  2.  Check the box next to Show verbose output during > ? compilation.
  3.  Check the box next to Show verbose output during > ? upload.
  4.  Click the OK** button.
  5.  Do an upload, just as you did before.
  6.  Wait for upload to fail.
  7.  After the upload fails you'll see a button on the "Upload error" notification at the bottom right corner of the IDE window: COPY ERROR MESSAGES. Click that button.
  8.  Paste<https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525> the error messages in a reply here.

-

per1234, 2022-09-16:
Thanks @supremeneuron. The only thing I can think of that would cause this error is that your installation of the "**Arduino SAMD Boards (32-bits ARM Cortex-M0+)**" platform of the "**Arduino Zero (Native USB Port)**" board you are uploading to has been modified or corrupted.

---

If you have modified it, then you will need to make sure your modified version contains the change made here: https://github.com/arduino/ArduinoCore-samd/commit/39f483e901a6c7c10cbacc67df2048502913cf8b

---

If you have not modified it, then the problem should be solved by uninstalling and then reinstalling the platform. I'll provide instructions you can follow to do that:

1. Select **Tools > Board > Boards Manager** from the Arduino IDE menus to open the "**Boards Manager**" view in the left side panel.
1. Scroll down through the list of boards platforms until you see the "**Arduino SAMD Boards (32-bits ARM Cortex-M0+)**" entry. Hover the mouse pointer over it.
1. Hover the mouse pointer over the "**INSTALLED**" label on the "**Arduino SAMD Boards (32-bits ARM Cortex-M0+)**" entry. You will now see it change to an "**UNINSTALL**" button.
1. Click the "**UNINSTALL**" button.
1. Wait for the uninstallation to finish.
1. Hover the mouse pointer over the "**Arduino SAMD Boards (32-bits ARM Cortex-M0+)**" entry again.
1. Click the "**INSTALL**" button.
1. Wait for the installation to finish.

Now try uploading again. Hopefully the error will no longer occur.

Please reply here to let us know the result.

supremeneuron, 2022-09-16:
Thank you so much!!! That worked!! I literally have no clue why everything worked with the older 1.8.x version but didn't with the new 2.0 version.  But uninstalling and re-installing the Board platform did the trick.

per1234, 2022-09-16:
You are welcome. I'm glad it is working now. Thanks for taking the time to report your results.

>I literally have no clue why everything worked with the older 1.8.x version

Arduino IDE 2.x uses a more flexible board definition format in order to support the new "pluggable discovery" system, which allows boards platform authors to use any arbitrary communication channel ("port") between the board and Arduino IDE, instead of assuming that there are only two possible port types: "serial" and "network" as the original design of Arduino IDE 1.x did.

This change was made in a backwards compatible manner, so that the old format board definitions still work just as well with Arduino IDE 2.x as they do with Arduino IDE 1.x. The board definitions of the "**Arduino SAMD Boards (32-bits ARM Cortex-M0+)**" boards platform have been updated to use the new format, while also keeping the properties recognized by Arduino IDE 1.x. Whatever the problem was in your previous "**Arduino SAMD Boards (32-bits ARM Cortex-M0+)**" boards platform only affected the properties of the new system. This is why Arduino IDE 1.x could still use it without any problem, but Arduino IDE 2.x could not.

In case anyone is interested in even more boring details, you can learn about it here:

https://arduino.github.io/arduino-cli/latest/platform-specification/#sketch-upload-configuration

closed 2022-09-16 by per1234