arduino/arduino-ide #1436: IDE wiped whole project folder including .git folder when renaming 1 file

benapetr, 2022-09-15:
### Describe the problem

On Windows I was renaming last remaining .ino file to .cpp (I didn't know project must have at least 1 .ino file to work) and IDE hanged up (mouse wheel rotating, IDE not responding). I killed it with task manager and to my surprise, whole project folder including .git folder got wiped from the FS - project folder was completely empty.

### To reproduce

Create project with many files, rename all .ino files to .cpp one by one

### Expected behavior

Not crash, not wipe project folder

### Arduino IDE version

2.0.0

### Operating system

Windows

### Operating system version

Windows 10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-09-20:
> I was renaming last remaining .ino file to .cpp

When renaming the main `.ino` file, whether this is the last one or not, the `Save As...` native file dialog comes up. Hence you're not renaming `YourSketch.ino` to `YourSketch.cpp` but saving `YourSketch` as something else inside the sketchbook. Would you mind sharing a screencast of what you did? I could not reproduce the defect locally so far.

> ### To reproduce
> Create project

Was your sketch already saved to the sketchbook or was it an _Unsaved_ one?

Thank you!

benapetr, 2022-10-01:
It was a sketch already created and saved by old Arduino IDE (1.x). I will try to reproduce this.

benapetr, 2022-10-02:
ok I can't reproduce it anymore either, maybe it was fixed very recently, now when I try to rename last .ino file it asks me to rename whole folder, before it asked just like when renaming file, then it got stuck and when I forced shut it down whole folder was gone.

closed 2022-10-02 by benapetr