arduino/arduino-ide #1437: "Check for Arduino IDE Updates" operation fails when automatic update checks are disabled

per1234, 2022-09-15:
### Describe the problem

On every startup, the Arduino IDE checks for new versions of the IDE and installed libraries and boards platforms. If a newer version of any of these is available, a notification/dialog is shown offering an update.

Although this is an important and convenient feature, it may be unwanted by some users. For example, the user may be intentionally using an outdated version of a library or boards platform either due to a bug or incompatibility introduced in the newest version, or else because they want to use the dependency versions their project is validated for. In this case, the non-actionable notifications for library/platform updates may be unwanted.

For this situation, an advanced setting (`arduino.checkForUpdates`) is provided to disable the update check (https://github.com/arduino/arduino-ide/pull/1361). The user may still wish to manually check for available updates for these components, which the Arduino IDE UI supports:

- IDE updates: **Help > Check for Arduino IDE Updates**
- Library updates: Set the "**Type**" filter in the Library Manager view to "**Updatable**"
- Platform updates: Set the "**Type**" filter in the Boards Manager view to "**Updatable**"

🐛 **Help > Check for Arduino IDE Updates** fails when `arduino.checkForUpdates` is set to `false`.

### To reproduce

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "**Command Palette**".
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
1. A "**Preferences**" tab will now open. Type `arduino.checkForUpdates` in the "**Search Settings**" field.
1. Uncheck the checkbox under the "**Check For Updates**" setting.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start the Arduino IDE.
1. Select **Help > Check for Arduino IDE Updates** from the Arduino IDE menus.

🐛 The update check fails with an error notification:

```text
Error while checking for Arduino IDE updates. Request checkForUpdates failed with message: 403 Forbidden "method: GET url: https://arduino-downloads-prod-beagle.s3.amazonaws.com/arduino-ide/nightly/latest.yml?noCache=1gd075653\n\n Data:\n <?xml version="1.0" encoding="UTF-8"?>\n<Error><Code>AccessDenied</Code><Message>Access Denied</Message><RequestId>JTE1MXVSAKE3N8X0</RequestId><HostId>xchwM0WmLnjIB5KD+lcQ+wh1p/aXcJfo+8pNx4yWOVaO8UdnKVNyODRjoVc/3Y73+N6Ju4wbWEk=</HostId></Error>\n " Headers: { "x-amz-request-id": "JTE1MXVSAKE3N8X0", "x-amz-id-2": "xchwM0WmLnjIB5KD+lcQ+wh1p/aXcJfo+8pNx4yWOVaO8UdnKVNyODRjoVc/3Y73+N6Ju4wbWEk=", "content-type": "application/xml", "transfer-encoding": "chunked", "date": "Thu, 15 Sep 2022 09:19:56 GMT", "server": "AmazonS3" }
```

### Expected behavior

**Help > Check for Arduino IDE Updates** is functional even when `arduino.checkForUpdates` is set to `false`.

### Arduino IDE version

2.0.0-snapshot-5f82577

### Operating system

Windows

### Operating system version

10

### Additional context

The logs shown when the error occurs:

```text
Checking for update
Error: HttpError: 403 Forbidden
"method: GET url: https://arduino-downloads-prod-beagle.s3.amazonaws.com/arduino-ide/nightly/latest.yml?noCache=1gd075653\n\n          Data:\n          <?xml version=\"1.0\" encoding=\"UTF-8\"?>\n<Error><Code>AccessDenied</Code><Message>Access Denied</Message><RequestId>JTE1MXVSAKE3N8X0</RequestId><HostId>xchwM0WmLnjIB5KD+lcQ+wh1p/aXcJfo+8pNx4yWOVaO8UdnKVNyODRjoVc/3Y73+N6Ju4wbWEk=</HostId></Error>\n          "
Headers: {
  "x-amz-request-id": "JTE1MXVSAKE3N8X0",
  "x-amz-id-2": "xchwM0WmLnjIB5KD+lcQ+wh1p/aXcJfo+8pNx4yWOVaO8UdnKVNyODRjoVc/3Y73+N6Ju4wbWEk=",
  "content-type": "application/xml",
  "transfer-encoding": "chunked",
  "date": "Thu, 15 Sep 2022 09:19:56 GMT",
  "server": "AmazonS3"
}
    at createHttpError (C:\ide 2\tip\125-5f82577\resources\app\node_modules\electron-updater\node_modules\builder-util-runtime\out\httpExecutor.js:14:12)
    at IncomingMessage.<anonymous> (C:\ide 2\tip\125-5f82577\resources\app\node_modules\electron-updater\node_modules\builder-util-runtime\out\httpExecutor.js:147:28)
    at IncomingMessage.emit (node:events:394:28)
    at endReadableNT (node:internal/streams/readable:1331:12)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
Request checkForUpdates failed with error: 403 Forbidden
"method: GET url: https://arduino-downloads-prod-beagle.s3.amazonaws.com/arduino-ide/nightly/latest.yml?noCache=1gd075653\n\n          Data:\n          <?xml version=\"1.0\" encoding=\"UTF-8\"?>\n<Error><Code>AccessDenied</Code><Message>Access Denied</Message><RequestId>JTE1MXVSAKE3N8X0</RequestId><HostId>xchwM0WmLnjIB5KD+lcQ+wh1p/aXcJfo+8pNx4yWOVaO8UdnKVNyODRjoVc/3Y73+N6Ju4wbWEk=</HostId></Error>\n          "
Headers: {
  "x-amz-request-id": "JTE1MXVSAKE3N8X0",
  "x-amz-id-2": "xchwM0WmLnjIB5KD+lcQ+wh1p/aXcJfo+8pNx4yWOVaO8UdnKVNyODRjoVc/3Y73+N6Ju4wbWEk=",
  "content-type": "application/xml",
  "transfer-encoding": "chunked",
  "date": "Thu, 15 Sep 2022 09:19:56 GMT",
  "server": "AmazonS3"
} HttpError: 403 Forbidden
"method: GET url: https://arduino-downloads-prod-beagle.s3.amazonaws.com/arduino-ide/nightly/latest.yml?noCache=1gd075653\n\n          Data:\n          <?xml version=\"1.0\" encoding=\"UTF-8\"?>\n<Error><Code>AccessDenied</Code><Message>Access Denied</Message><RequestId>JTE1MXVSAKE3N8X0</RequestId><HostId>xchwM0WmLnjIB5KD+lcQ+wh1p/aXcJfo+8pNx4yWOVaO8UdnKVNyODRjoVc/3Y73+N6Ju4wbWEk=</HostId></Error>\n          "
Headers: {
  "x-amz-request-id": "JTE1MXVSAKE3N8X0",
  "x-amz-id-2": "xchwM0WmLnjIB5KD+lcQ+wh1p/aXcJfo+8pNx4yWOVaO8UdnKVNyODRjoVc/3Y73+N6Ju4wbWEk=",
  "content-type": "application/xml",
  "transfer-encoding": "chunked",
  "date": "Thu, 15 Sep 2022 09:19:56 GMT",
  "server": "AmazonS3"
}
    at createHttpError (C:\ide 2\tip\125-5f82577\resources\app\node_modules\electron-updater\node_modules\builder-util-runtime\out\httpExecutor.js:14:12)
    at IncomingMessage.<anonymous> (C:\ide 2\tip\125-5f82577\resources\app\node_modules\electron-updater\node_modules\builder-util-runtime\out\httpExecutor.js:147:28)
    at IncomingMessage.emit (node:events:394:28)
    at endReadableNT (node:internal/streams/readable:1331:12)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
```

---

The issue does not occur when `arduino.checkForUpdates` is set to `true`.

---

Originally reported at https://forum.arduino.cc/t/arduino-ide-2-0-0-available-for-download/1031651/5 by @sterretjeToo

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-09-27:
Can we close this @AlbyIanna? See https://github.com/arduino/arduino-ide/pull/1490.

AlbyIanna, 2022-09-27:
Sure!

closed 2022-09-27 by AlbyIanna