arduino/arduino-ide #1438: Network issue and configuration

fpistm, 2022-09-15:
### Describe the problem

Hi,

I've installed the new official release 2.0.0 (on Windows 10 and Linux Ubuntu 22.04 LTS) from archive.

When launching it, I got this windows and nothing happen.
[image]

## First step:
Log from the console are (Linux):

> Starting backend process. PID: 667034
[667055:0915/102131.293101:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///local/home/<username>/.theia'
root INFO Backend Object.initialize: 7.5 ms [Finished 1.253 s after backend start]
root INFO Backend Object.configure: 5.8 ms [Finished 1.253 s after backend start]
root INFO Backend MessagingContribution.onStart: 2.9 ms [Finished 1.253 s after backend start]
root INFO Backend OSBackendApplicationContribution.configure: 0.7 ms [Finished 1.320 s after backend start]
root INFO Backend DefaultWorkspaceServer.onStart: 0.6 ms [Finished 1.320 s after backend start]
root INFO Backend LocalizationBackendContribution.configure: 0.4 ms [Finished 1.321 s after backend start]
root WARN Backend LocalizationBackendContribution.initialize took longer than the expected maximum 50 milliseconds: 67.7 ms [Finished 1.321 s after backend start]
root INFO Backend TaskBackendApplicationContribution.onStart: 0.6 ms [Finished 1.321 s after backend start]
root INFO Backend ElectronTokenBackendContribution.configure: 1.7 ms [Finished 1.323 s after backend start]
root INFO Backend PluginDeployerContribution.initialize: 0.8 ms [Finished 1.323 s after backend start]
root INFO Backend ConfigServiceImpl.onStart: 1.4 ms [Finished 1.325 s after backend start]
root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.7 ms [Finished 1.326 s after backend start]
root INFO Backend ArduinoDaemonImpl.onStart: 1.6 ms [Finished 1.327 s after backend start]
root INFO Backend NodeFileUploadService.configure: 3.0 ms [Finished 1.328 s after backend start]
daemon INFO Starting daemon from /local/data/ide/arduino/arduino_2.0.0/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
discovery-log INFO start
discovery-log INFO start new deferred
root INFO Backend BoardDiscovery.onStart: 1.6 ms [Finished 1.330 s after backend start]
root INFO Backend FileDownloadEndpoint.configure: 0.8 ms [Finished 1.330 s after backend start]
root INFO Backend AuthenticationServiceImpl.onStart: 0.6 ms [Finished 1.332 s after backend start]
root INFO Backend PluginApiContribution.configure: 0.7 ms [Finished 1.332 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.4 ms [Finished 1.333 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.1 ms [Finished 1.334 s after backend start]
root INFO Backend PlotterBackendContribution.configure: 0.5 ms [Finished 1.334 s after backend start]
root INFO Theia app listening on http://localhost:40205.
root INFO Finished starting backend application: 5.5 ms [Finished 1.338 s after backend start]
root WARN The local plugin referenced by local-dir:/local/home/<username>/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/local/home/<username>/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:/local/home/<username>/.arduinoProIDE/plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.5" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-cs" to a VS Code extension "vscode-language-pack-cs@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-es" to a VS Code extension "vscode-language-pack-es@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-it" to a VS Code extension "vscode-language-pack-it@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-ja" to a VS Code extension "vscode-language-pack-ja@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-nl" to a VS Code extension "vscode-language-pack-nl@1.48.3" with engines:
[... 149 lines cut ...]

### Expected behavior

- Warn user properly instead of having empty windows with 'splashscreen' or proposed to configure network proxy.
- Document proxy settings requirements for Arduino IDE 2.0.0

Maybe partially linked to #29

### Arduino IDE version

2.0.0

### Operating system

Windows, Linux

### Operating system version

Windows10 and Linux ubuntu 22.04 LTS

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-09-27:
TODOs on the IDE2 side:
 - [ ] Search for a JS lib to detect the system proxy configuration. The candidate is [`os-proxy-config`](https://www.npmjs.com/package/os-proxy-config) from [HTTP Toolkit](https://github.com/httptoolkit).
 - [ ] The core gRPC client initialization must be resilient. Even on a fatal error, the IDE2 should open.
 - [ ] IDE2 must start even if no platforms/libs were downloaded the very first time IDE2 starts. It's OK if the boards/lib manager is not functional, but the users must be notified when there is no Internet connection or the CLI cannot download anything due to being behind a proxy.
    - Preferably, the proxy configuration should be adjusted from the _Settings_ dialog by users.
    - If the _Settings_ dialog is not functional, IDE2 must open the `~/arduinoIDE/arduino-cli.yaml` configuration file inside IDE2 for editing.

TODOs on the CLI side:
 - [ ] In the long run, `network#proxy#'auto'` should be supported.
 - [ ] Search for a Golang lib that can detect the system proxy configuration.

tennessine, 2023-01-18:
Same issue on macos Monterey 12.6.3

arduino ide 2.0.4 nightly

Lucius0806, 2023-03-09:
What is a proxy server login? <proxy scheme>: the URL scheme for your proxy server.

DL7BJ, 2023-06-06:
Same issue with Arduino IDE 2.1.0 for Windows.

Dmitry-Chernikov, 2023-08-23:
Hello, according to the information [from here](https://developer.mozilla.org/en-US/docs/Web/HTTP/Proxy_servers_and_tunneling/Proxy_Auto-Configuration_PAC_file) Proxy Auto-Configuration file is a js script with FindProxyForURL, Node.js function executes this script from the command line. You can get the result of the execution and substitute it automatically into the function that uses the settings:
network:
   proxy: http://<username>:<password>@<proxy>:<port>

scls19fr, 2023-09-18:
I can't use Arduino 2.x at work (university) with my students until this is fixed.
I really hope to see soon a fix about it!

fpistm, 2023-09-18:
> I can't use Arduino 2.x at work (university) with my students until this is fixed. I really hope to see soon a fix about it!

With the workaround, it should works even if it is not user friendly.

One thing we noticed is we update the IDE thanks the update notification, the network configuration set in the arduino-cli.yaml is not preserved.