arduino/arduino-ide #1439: Same binary name across host OS

fpistm, 2022-09-15:
### Describe the request

Hi,

I would recommend to avoid space in the binary name and use the same name across platform.
This would ease usage mainly in console.

### Describe the current behavior

On Windows, executable name is `Arduino IDE` while on Linux it is `arduino-ide`.
I have not checked on Mac.

### Arduino IDE version

2.0.0

### Operating system

Windows, Linux

### Operating system version

NA

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

kittaakos, 2022-09-15:
> On Windows, executable name is `Arduino IDE` while on Linux it is `arduino-ide`.
> I have not checked on Mac.

The expected is `Arduino IDE`.

> This would ease usage mainly in console.

If you can use a terminal, I expect that escaping the space is no problem.

> while on Linux it is `arduino-ide`.

Maybe you're looking for this change: https://github.com/arduino/arduino-ide/pull/356/. Once there is an app launcher for IDE2 on Linux, the app name will be consistent on all operating systems. See the proposal [here](https://github.com/arduino/arduino-ide/pull/356/files#diff-470063f7963ac419c558f13c066ab4853c462227bb679334f5d555b138405d64R4).

fpistm, 2022-09-15:
> The expected is `Arduino IDE`.

So why not the same on Linux :)

> If you can use a terminal, I expect that escaping the space is no problem.

I know but really not convenient.

> Maybe you're looking for this change: #356. Once there is an app launcher for IDE2 on Linux, the app name will be consistent on all operating systems. See the proposal [...]

This is not the binary name just an alias or icone name.