arduino/arduino-ide #1447: Impossible to ship translation contributions for many languages

per1234, 2022-09-16:
### Describe the problem

Arduino IDE 1.x has been [localized](https://en.wikipedia.org/wiki/Internationalization_and_localization) to 66 languages thanks to the amazing work of the community. Because Arduino IDE 2.x is a complete software rewrite, it uses a separate [internationalization](https://en.wikipedia.org/wiki/Internationalization_and_localization) system.

The approach taken in this project is to leverage the internationalization infrastructure and localization data that already exists for [VS Code](https://code.visualstudio.com/). These assets are applicable to Arduino IDE 2.x due to the project being built on the [**Eclipse Theia** IDE framework](https://theia-ide.org/). In this way, the Arduino IDE developers and community are only responsible for internationalization and localization of the Arduino-specific strings (e.g., "Sketchbook") used in the IDE's UI.

The downside to this approach is that the localization of the project is limited to [languages for which a VS Code "language pack" is available](https://open-vsx.org/?search=&category=Language%20Packs).

It is Arduino's responsibility to ensure the generous contributions of volunteer translators are brought to fruition by shipping Arduino IDE 2.x releases localized to that language.

🐛 Arduino IDE 2.x translations for a significant number of languages without language packs have been created. It is not clear if or how it will be possible to ship all these translations:

| Languages on<br />Transifex | Language pack<br />available?                                                 |
| --------------------------- | ----------------------------------------------------------------------------- |
| Afrikaans                   | ❌                                                                            |
| Arabic                      | ❌                                                                            |
| Azerbaijani                 | ❌                                                                            |
| Basque                      | ❌                                                                            |
| Bulgarian                   | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-bg)        |
| Burmese (Myanmar)           | [✔️](https://open-vsx.org/extension/MyoHlaingWin/vscode-language-pack-mm-uni) |
| Catalan (Spain)             | ❌                                                                            |
| Chinese                     | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-zh-hans)   |
| Chinese (Taiwan)            | [❔](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-zh-hant)   |
| Czech                       | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-cs)        |
| Dutch                       | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-nl)        |
| Filipino                    | ❌                                                                            |
| French                      | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-fr)        |
| German                      | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-de)        |
| Greek                       | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-hu)        |
| Hebrew                      | ❌                                                                            |
| Hungarian                   | ❌                                                                            |
| Italian                     | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-it)        |
| Japanese                    | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-ja)        |
| Korean                      | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-ko)        |
| Nepali                      | ❌                                                                            |
| Persian                     | ❌                                                                            |
| Polish                      | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-pl)        |
| Portuguese                  | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-pt-BR)     |
| Romanian                    | ❌                                                                            |
| Russian                     | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-ru)        |
| Serbian                     | ❌                                                                            |
| Spanish                     | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-es)        |
| Turkish                     | [✔️](https://open-vsx.org/extension/MS-CEINTL/vscode-language-pack-tr)        |
| Vietnamese                  | ❌                                                                            |

### To reproduce

1. Check the list of languages in the Arduino IDE 2.x Transifex project:
   https://explore.transifex.com/arduino-1/ide2/
1. Select **File > Preferences** from the Arduino IDE menus.
1. Open the "**Language**" menu.
   🐛 Only a fraction of the translations have been shipped
1. Check the list of available VS Code language packs:
   https://open-vsx.org/?search=&category=Language%20Packs

🐛 Language packs are unavailable for a significant number of the languages which have Arduino IDE 2.x translation initiatives in progress.

### Expected behavior

Arduino is prepared to ship community contributions as soon as a significant localization coverage has been reached.

### Arduino IDE version

2.0.0-snapshot-4e590ab

### Operating system

All

### Additional context

In addition to the immediate problem that shipping existing contributions will either be delayed or not feasible, there is also the issue of the limitations on being able to add additional languages (e.g., [here we cannot accept an offer of Galician translation](https://forum.arduino.cc/t/arduino-ide-2-0-0-available-for-download/1031651/8)), meaning that the localization coverage standard set by Arduino IDE 1.x can not be met.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

bohrsty, 2022-09-17:
just a question, because it fits to the topic:
in the transifex list the german translation is completed (100%), but if you open the ide with german language, i.e. the "edit" menu isn't translated, it is shown as "Edit" in english... same here inside the menu with "copy" and "paste", they are shown in english too (maybe more i haven't found yet)...

what is the problem here and where could we support translate it completely and correctly?

per1234, 2022-09-17:
Hi @bohrsty. That bug is tracked here: https://github.com/arduino/arduino-ide/issues/1431
(The Chinese language is used as an example, but it affects all languages)

I'm happy to tell you that a fix is already in progress: https://github.com/arduino/arduino-ide/pull/1449

bohrsty, 2022-09-17:
thanks for the info, i appreciate for your work and will follow the issue for further progress...

g-castronovo, 2022-10-26:
Hello !_! Any news on this? I have got requests for more languages but I've not yet accepted them. Keep me in the loop <3

per1234, 2022-11-03:
@ubidefeo @davegarthsimpson can you provide us with an update on this? ^^^