arduino/arduino-ide #1454: Version conflict between Arduino IDE 2.0.0 from arduino.cc and from Microsoft Windows Package Manager (winget) repository

mscott370, 2022-09-17:
### Describe the problem

After installing the latest /STABLE) Arduino IDE 2.0.0 from arduino.cc/en/software on 16/09, i checked  to see if this product was supported by Microsoft's Package Manager ("Winget") and was surprised to be offered an update:  
PS C:\Users\Mike> winget list --name "Arduino IDE 2.0.0"
Name              Id                 Version   Available Source
---------------------------------------------------------------
Arduino IDE 2.0.0 ArduinoSA.IDE.beta < 2.0.0.0 2.0.0.0   winget
PS C:\Users\Mike> winget show --id "ArduinoSA.IDE.beta"
Found Arduino IDE Beta [ArduinoSA.IDE.beta]
Version: 2.0.0.0
Publisher: Arduino SA
Moniker: arduino-beta
Description: The open-source Arduino Software (IDE) makes it easy to write code and upload it to the board. This software can be used with any Arduino board.
License: GNU LGPL
Installer:
  Type: wix
  Download Url: https://downloads.arduino.cc/arduino-ide/arduino-ide_2.0.0-beta.12_Windows_64bit.msi
  SHA256: 3f4c0ff9a3273e764fc61e409be2e23540bded8257d4c252fdb5033ce676376d

The Package Manager app Winget regards this 2.0.0 beta version12 (created 30/11/2021) as being higher / more recent than the 2.0.0 stable version from arduino.cc which was created on 14/09/2022 and happily updates to the back-level beta version!
Unfortunately, although the beta version installs without problems, it will not start, so the only way forward is to uninstall the beta version and then re-install the 2.0.0 **stable** version from arduino.cc. 

### To reproduce

Download latest Arduino IDE 2.0.0 version dated 14.09.2022 from arduino.cc/../software and install it under Windows 10. 
Then invoke winget command to see if any updates are available in the package repository:
_winget list --name "Arduino IDE 2.0.0"_
You should see that a potential update candidate is offered as above.

### Expected behavior

I would expect that the (older) beta release is not considered suitable as an upgrade - even though both versions are "2.0.0"

### Arduino IDE version

2.0.0 (stable)

### Operating system

Windows

### Operating system version

Windows 10 Home Version 22H2, Build 19045.2006

### Additional context

Microsoft apparently intends to integrate the "WINGET" tool into the Microsoft Store, to make update checking easier for windows users. The currently out-of-date content in Microsoft's package repository makes this tool a potential source of problems.
I don't know how the repository gets updated, but I guess the responsibility for providing the latest versions lies with the software creators like Arduino LLC / Arduino SA.
Please clarify the cause of this conflict.
P.S. I have intentionally **not** installed a "nightly build" because that would merely complicate this issue, and i would like to see the conflict resolved for the **Stable** version..

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kadun, 2022-09-26:
Please update winget package to the last release, still upgrading (downgrading actually) to beta version

per1234, 2022-09-26:
Hi @mscott370 and @kadun. The Arduino IDE packages in the Windows Package Manager/winget repository are something that was submitted by the community without any involvement from Arduino:

https://github.com/microsoft/winget-pkgs/commits/master/manifests/a/ArduinoSA/IDE

It is recommended to always download the Arduino IDE from the "**Software**" page on arduino.cc:

https://www.arduino.cc/en/software/

That, and the builds found in this repository are the only official distributions of the Arduino IDE. All other distributions are maintained by the community. If you find a problem with one of the alternative distributions then you should report that to the 3rd party maintainers responsible for it, or submit a fix to the relevant repository.

closed 2022-09-26 by per1234