arduino/arduino-ide #1456: Library manager does not work

VlaJ123, 2022-09-18:
### Describe the problem

The library manager does not show anything.
[image]

### To reproduce

Open the library manager and search for anything, then press enter.

### Expected behavior

It should have shown available libraries.

### Arduino IDE version

arduino-ide_nightly-20220918_Windows_64bit

### Operating system

Windows

### Operating system version

Windows 11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-18:
Hi @VlaJ123. Thanks for your report.

I'm going to ask you to provide some additional information that might help us to identify the problem.

Please do this:

1. Start Windows "**File Explorer**".
1. Open the Arduino IDE installation folder
   (usually `C:\Program Files\Arduino IDE` or `C:\Users\<user name>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the **<kbd>Shift</kbd>** key while clicking the right hand button on the mouse.
1. From the context menu, click "**Open PowerShell window here**".
   PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type the following command:
   ```text
   & ".\Arduino IDE"
   ```
1. Press the **<kbd>Enter</kbd>** key.
   Debug output should now be printed to the PowerShell window as the Arduino IDE starts (or fails to start).
1. Wait for the Arduino IDE to finish starting.
1. Switch back to the PowerShell window if the Arduino IDE window took the focus.
1. Right click on the title bar of the PowerShell window.
   This will open a context menu.
1. Select **Edit > Select All** from the context menu.
   This will select all the text in the PowerShell window.
1. Press **<kbd>Ctrl</kbd>+<kbd>C</kbd>**.
   This will copy the contents of the PowerShell window to the clipboard.
1. Paste (**<kbd>Ctrl</kbd>+<kbd>V</kbd>**) the output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

VlaJ123, 2022-09-18:
Hello, the debug output is

`Microsoft Windows [Version 10.0.22000.978]
(c) Microsoft Corporation. All rights reserved.

C:\Users\vladr\Downloads\arduino-ide_nightly-20220918_Windows_64bit>".\Arduino IDE"

C:\Users\vladr\Downloads\arduino-ide_nightly-20220918_Windows_64bit>
Starting backend process. PID: 42484
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///c%3A/Users/vladr/.theia'
root INFO Backend Object.initialize: 4.3 ms [Finished 4.073 s after backend start]
root INFO Backend Object.configure: 3.5 ms [Finished 4.073 s after backend start]
root INFO Backend MessagingContribution.onStart: 1.4 ms [Finished 4.073 s after backend start]
root INFO Backend OSBackendApplicationContribution.configure: 0.3 ms [Finished 4.265 s after backend start]
root INFO Backend DefaultWorkspaceServer.onStart: 0.3 ms [Finished 4.265 s after backend start]
root INFO Backend LocalizationBackendContribution.configure: 0.2 ms [Finished 4.265 s after backend start]
root WARN Backend LocalizationBackendContribution.initialize took longer than the expected maximum 50 milliseconds: 192.0 ms [Finished 4.265 s after backend start]
root INFO Backend TaskBackendApplicationContribution.onStart: 0.2 ms [Finished 4.265 s after backend start]
root INFO Backend ElectronTokenBackendContribution.configure: 1.2 ms [Finished 4.267 s after backend start]
root INFO Backend PluginDeployerContribution.initialize: 0.4 ms [Finished 4.267 s after backend start]
root INFO Backend ConfigServiceImpl.onStart: 0.8 ms [Finished 4.268 s after backend start]
root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.3 ms [Finished 4.268 s after backend start]
root INFO Backend ArduinoDaemonImpl.onStart: 0.9 ms [Finished 4.269 s after backend start]
root INFO Backend NodeFileUploadService.configure: 1.8 ms [Finished 4.269 s after backend start]
daemon INFO Starting daemon from C:\Users\vladr\Downloads\arduino-ide_nightly-20220918_Windows_64bit\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
discovery-log INFO start
discovery-log INFO start new deferred
root INFO Backend BoardDiscovery.onStart: 0.7 ms [Finished 4.271 s after backend start]
root INFO Backend FileDownloadEndpoint.configure: 0.5 ms [Finished 4.271 s after backend start]
root INFO Backend AuthenticationServiceImpl.onStart: 0.3 ms [Finished 4.272 s after backend start]
root INFO Backend PluginApiContribution.configure: 0.5 ms [Finished 4.272 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.4 ms [Finished 4.273 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.1 ms [Finished 4.273 s after backend start]
root INFO Backend PlotterBackendContribution.configure: 3.9 ms [Finished 4.278 s after backend start]
root INFO Theia app listening on http://localhost:58782.
root INFO Finished starting backend application: 7.7 ms [Finished 4.280 s after backend start]
root WARN The local plugin referenced by local-dir:/c%3A/Users/vladr/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/vladr/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\vladr\.arduinoProIDE\plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.5" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-cs" to a VS Code extension "vscode-language-pack-cs@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-es" to a VS Code extension "vscode-language-pack-es@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-it" to a VS Code extension "vscode-language-pack-it@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-ja" to a VS Code extension "vscode-language-pack-ja@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-nl" to a VS Code extension "vscode-language-pack-nl@1.48.3" with engines:
[... cut ...]
daemon INFO {"level":"info","msg":"from discovery builtin:mdns-discovery received message type: start_sync, message: OK","time":"2022-09-18T21:25:14+03:00"}

discovery-log INFO {"eventType":"add","port":{"matchingBoardsList":[],"port":{"address":"COM1","label":"COM1","protocol":"serial","protocolLabel":"Serial Port","propertiesMap":[]}},"error":""}
discovery-log WARN Port 'COM1 Serial Port' was already available. Known boards before override: []
root INFO Attached boards and available ports changed:
root INFO No changes.
root INFO ------------------------------------------

`

per1234, 2022-09-18:
The cause is shown by this error message:
>```text
>daemon INFO ERROR: 2022/09/18 21:24:47 [core] grpc: server failed to encode response:  rpc error: code = Internal desc = grpc: error while marshaling: string field contains invalid UTF-8
>
>root ERROR Request search failed with error: 13 INTERNAL: grpc: error while marshaling: string field contains invalid UTF-8
>```

There is an unfortunate bug in Arduino IDE or Arduino CLI that causes this to happen when you have a library installed with character encoding other than UTF-8 in its metadata file.

This bug is tracked here:

https://github.com/arduino/arduino-ide/issues/135

The workaround is to convert the encoding of the metadata file to UTF-8. That is easy to do with any decent text editor (e.g., [VS Code](https://code.visualstudio.com/), [Notepad++](https://notepad-plus-plus.org/)) once you have identified the problematic library. Unfortunately the error message does not tell us which library or libraries it is.

It will be one of the `library.properties` files directly under the subfolders of this path:

```text
c:\Users\vladr\OneDrive\Documents\Arduino\libraries\
```

You can open each of the files in turn using a text editor that shows the encoding. Or if you prefer you can use a "bisection" technique by moving half of the libraries out of the folder, restarting the Arduino ide, then checking whether or not the issue still occurs. If not, you know the problematic library was in the half removed. Repeating that process will narrow it down to the problematic library or libraries.

---

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

If you would like further assistance with the workaround for the bug so that you can get the Arduino IDE into a fully functional state, please post over on the Arduino forum. I'll be sure to see your post over there and help you out:

https://forum.arduino.cc/c/software/arduino-ide-2-0/93

closed 2022-09-18 by per1234