arduino/arduino-ide #1458: Possibility to don't update selected libraries

Cyb3rn0id, 2022-09-18:
### Describe the request

I'm using some old libraries (not important which) since updated ones doesn't work. 
So, everytime Arduino IDE 2 starts, it notifies me about updated libraries and then everytime I select "INSTALL ALL" since is good to update libraries, right? But this will cause updating of THAT library I want the old version. For me will be good something like a flag for every installed libraries saying "Don't notify me for updates about this" or something similar.

### Describe the current behavior

There is no option for a quick update don't regarding some selected libraries I don't want to update

### Arduino IDE version

2.0.0

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-09-18:
Hi @Cyb3rn0id. Thanks for your suggestion.

Although there isn't a granular control as you request, it is possible to disable the update checks entirely in the [advanced settings](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings), which might be considered at least an improvement in a situation like this even if it doesn't fulfill your request.

I'll provide instructions:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "**Command Palette**".
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
1. A "**Preferences**" tab will now open. Type `arduino.checkForUpdates` in the "**Search Settings**" field.
1. Uncheck the box under the "**Check For Updates**" setting.
1. Click the **X** icon on the "**Preferences**" tab to close it.

After this, you can still check for updatable libraries and boards platforms by opening the Library Manager or Boards Manager views and then selecting "**Updatable**" from the "**Type**" filter.

It should also be possible to check for Arduino IDE updates by selecting **Help > Check for Arduino IDE Updates** from the Arduino IDE menus, but there is currently a bug that prevents this: https://github.com/arduino/arduino-ide/issues/1437. So for now, just make sure to periodically check [the "**Software**" page on arduino.cc](https://www.arduino.cc/en/software/) to see if there is a new version of the Arduino IDE available.

kwit98, 2023-08-17:
Hi, 

Just copy whole library (in the same places where oryginals files are) and change name of it. Don't make random names, u'll forget what they mean. 

I didn't had any problems with new library, so i'm not sure if i copied all files, but after doing this I updated all libraries and uploaded my code once again. My mcu printed to serial monitor version of library and it was old (saved) one. 

// it's propobly too late response for u but this thread is one of the first showing in google, so maybe someone else will use it.