arduino/arduino-ide #1461: Vertical Tabs or File Explorer

Silver-Fang, 2022-09-19:
### Describe the request

When the project contains a lot of code files, I want them to be arranged vertically on the left side of the window. Better if it could support folder level expansion or file explorer. (Yeah, just like VS Code.)
[image]

### Describe the current behavior

A large number of files currently pile up into horizontal tabs at the top, or even shrink into menus.
[image]

### Arduino IDE version

2.0.0

### Operating system

Windows

### Operating system version

Windows 11

### Additional context

At present, I still rely on VS Code for development, and the Arduino IDE is only used for uploading code. Hopefully someday the Arduino IDE will evolve into a really good enough IDE that I don't need to open VS Code anymore.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-09-19:
>Better if it could support folder level expansion

Hi @Silver-Fang. I'm happy to tell you that the IDE already supports this. It is not enabled by default, so you will need to enable it in the preferences. I'll provide instructions:

1. Select **File > Preferences...** from the Arduino IDE menus.
1. Check the box next to "**☐ Show files inside Sketches**"
1. Click the <kbd>**OK**</kbd> button.
1. Open the "**Sketchbook**" view by clicking the folder icon on the Arduino IDE activity bar.

You will see that the sketchbook folders listed there have **❯** controls that allow expanding them. You can now navigate the files of the open sketch by clicking the target file in the "**Sketchbook**" view.

closed 2022-09-19 by per1234

Silver-Fang, 2022-09-19:
> > Better if it could support folder level expansion
> 
> Hi @Silver-Fang. I'm happy to tell you that the IDE already supports this. It is not enabled by default, so you will need to enable it in the preferences. I'll provide instructions:
> 
> 1. Select **File > Preferences...** from the Arduino IDE menus.
> 2. Check the box next to "**☐ Show files inside Sketches**"
> 3. Click the **OK** button.
> 4. Open the "**Sketchbook**" view by clicking the folder icon on the Arduino IDE activity bar.
> 
> You will see that the sketchbook folders listed there have **❯** controls that allow expanding them. You can now navigate the files of the open sketch by clicking the target file in the "**Sketchbook**" view.

No, this can only show a fixed folder, which is %USERPROFILE%\Documents\Arduino by default. I want to show folders of the sketch currently opened, even if it is not under the default directory.