arduino/arduino-ide #1463: Return to the position before "Go to definition"

Silver-Fang, 2022-09-19:
### Describe the request

Sometimes I need to go deep down a chain of definitions, and then need to go back, only to find that there is no option to go back, which is annoying.

### Describe the current behavior

There is currently no button to return to the previous position.

### Arduino IDE version

2.0.0

### Operating system

Windows

### Operating system version

Windows 11

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

Willem43T, 2022-09-20:
@Silver-Fang Thanks for submitting that request.

The "Go to Definition" is a hugely useful feature. I too, however, find no option to return to the previous position a big shortcoming.

r3inbowari, 2022-09-29:
hi, `alt+←` may be helpful to you.

closed 2022-09-30 by per1234