arduino/arduino-ide #1466: IDE 2 Asking for authentication

cjdg, 2022-09-19:
### Describe the problem

On startup, there is a unlock login keyring window asking for a password, but none of the passwords i use work, so dont know whats going here..

### To reproduce

Install flatpak version
Run Arduino IDE 2

### Expected behavior

No password  requiring, or at least should work using the valid ones

### Arduino IDE version

2

### Operating system

Linux

### Operating system version

22.04

### Additional context

Flatpak version

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-19:
Hi @cjdg. Please try the official Arduino IDE package downloaded from the links on arduino.cc:

https://www.arduino.cc/en/software

Once you have done that, please comment here to let me know whether or not the problem still occurs.

cjdg, 2022-09-20:
I already tested zip and AppImage and no problem with them, something is wrong with the flatpak

per1234, 2022-09-20:
Thanks for the update. The ["Arduino IDE v2" Flatpak package](https://github.com/flathub/cc.arduino.IDE2) is a 3rd party project and Arduino does not provide support for it via this repository.

If you still want to use the Flatpak, you can ask for assistance over on the Arduino Forum:

https://forum.arduino.cc/

Or maybe the Flathub forum would be more knowledgeable about this specific subject matter?

https://discourse.flathub.org/

closed 2022-09-20 by per1234

joshuag1000, 2023-03-01:
> Thanks for the update. The ["Arduino IDE v2" Flatpak package](https://github.com/flathub/cc.arduino.IDE2) is a 3rd party project and Arduino does not provide support for it via this repository.
>
> If [...]

Hi, @per1234 

Thanks for triaging flatpak issues. I try to regularly check here to ensure that the flatpak version that i maintain isn't causing any issues for you guys here.
Recently i have really busy and with no way to setup an automatic email filter for issues i havent been able to check.
To reduce any further workload on you guys I'm going to change the flathub issue url to the flathub issues at https://github.com/flathub/cc.arduino.IDE2/issues to ensure any future issues are dealt by me before passing them to your team if necessary. I'm also going to make an unofficial wrapper notice on the flathub page for the IDE mentioning that it is an unofficial wrapper so people are aware.
In the event an issue slips through would it be possible if you could ping/tag me @joshuag1000 before you close the issue so github will send me an email and i can work on fixing the issue that would be amazing!

Apologies for the inconvenience caused.
- J