arduino/arduino-ide #1473: Output view does not correctly handle carriage return

per1234, 2022-09-20:
### Describe the problem

Some command line tools generate an animated progress bar by using a carriage return without newline to repeatedly overwrite the same line.

🐛 This does not work when their output is printed in the "**Output**" view. The carriage return causes a line break to be added.

### To reproduce

1. Upload a sketch via the Arduino CLI command line interface to a board that uses an upload tool that produces an animated progress bar (e.g., Nano 33 BLE).
   🙂 There is a nice looking animated progress bar:
   [image]
1. Do the same using Arduino IDE 1.8.19.
   🙂 There is a nice looking animated progress bar:
   [image]
1. Do the same using Arduino IDE 2.x
   🐛 The progress bar is not animated:
   [image]

### Expected behavior

The current line is overwritten from the start of the line in the "Output" view with subsequent text.

For example, this text:

```text
foobar\rbaz
```

Should result in this content in the output view:

```text
bazbar
```

### Arduino IDE version

2.0.0-snapshot-4e590ab

### Operating system

Windows, Ubuntu

### Operating system version

Windows 10, Ubuntu 20.04

### Additional context

Support was added in Arduino IDE 1.x by https://github.com/arduino/Arduino/pull/9954

---

Originally reported at https://forum.arduino.cc/t/arduino-2-0-0-mac-ugh-why-so-backwards/1033549

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-20:
Thank @cmaglie. I didn't interpret the gRPC response content correctly. You are correct that Arduino CLI is handling it correctly and this is a bug in Arduino IDE. I have updated the issue and transferred it to the correct repo.

kittaakos, 2022-09-21:
The _Output_ view is an editor and not a terminal. If we want terminal behavior, we must use a [`Pseudoterminal`](https://code.visualstudio.com/api/references/vscode-api#Pseudoterminal) from VS Code. This change requires a bigger effort on the IDE2 side.

NoSloppy, 2025-02-06:
Sorry to necro this, but I don't see where the issue is at currently. Moved to Arduino IDE issues? or...?
Link anyone? Thanks.