arduino/arduino-ide #1482: cpu always 100% on macOS

pidloop, 2022-09-21:
### Describe the problem

cpu is always 100% on macOS

### To reproduce

just start the program, don't have to do anything

### Expected behavior

would expect cpu to be very low if I'm not doing anything

### Arduino IDE version

2.0.0

### Operating system

macOS

### Operating system version

Monterey 12.5.1

### Additional context

1.8.16 doesn't do this.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-22:
Hi @pidloop. Thanks for your report.

Which board (or boards) do you have selected in the Arduino IDE when this issue occurs?

Does the issue occur when you are editing a specific sketch (or sketches), or does it occur even with the basic sketch you get from **File > New**?

pidloop, 2022-09-22:
Hi, thanks for getting back.

1. Adafruit Feather HUZZAH ES8266

2. Just my sketch. If I open File -> New; select the same board; close my
sketch window; then the CPU goes back down to near zero (once the indexing
is finished).

Hmm, I just noticed something. With my sketch open and CPU is pegged, if I
minimize my sketch's window (macOS yellow window button) then the CPU also
goes down to zero. As soon as I open it back up then CPU pegs again. This
is entirely repeatable. Some sort of GUI issue perhaps?

BTW: my sketch is pretty complex: 68 files, 170 KLOC. I started it five
years ago -- you can see it here
<https://clearskyinstitute.com/ham/HamClock>.

per1234, 2022-09-22:
Very cool project!

I tested it on my Windows machine (I don't have a macOS machine) and was not able to reproduce the problem. The CPU usage is very high during the language server processing (which is associated with the "indexing" status sometimes shown at the left side of the status bar) after opening the sketch, but that CPU usage doesn't change if I minimize the window and it also does subside after approximately 1 minute when the processing finishes.

I'm going to request you to provide some additional information that will allow us to identify which specific process is using the CPU. I'll suggest two alternative options. You can pick whichever one of the two is more convenient for you (some people prefer a GUI, others a command line):

### GUI

Please provide a screenshot of the "**Activity Monitor**" window with all of the relevant processes visible:

1. Open the ["**Activity Monitor**"](https://support.apple.com/en-ca/guide/activity-monitor/actmntr1001/mac) application.
1. Select "**CPU**" from the toolbar.
1. Right click on the heading row of the process list.
1. If it is not already selected, click "**Memory**" in the dropdown menu to add that column to the list.
1. Select **View > All Processes, Hierarchically** from the menu.
1. Make sure all subprocesses under the "**Arduino IDE**" process are shown by clicking the **❯** icon at the left of the processes.
1. Make sure all the relevant processes are visible in the window.
1. Make sure all the relevant columns are visible in the window:
   - "**Process Name**"
   - "**% CPU**"
   - "**Memory**"
1. Take a [screenshot](https://support.apple.com/en-us/HT201361) of the "**Activity Monitor**" window.

Paste or attach the screenshot in a comment here. Please note that you must come to the GitHub issue web page to make an attachment:

https://github.com/arduino/arduino-ide/issues/1482

Attachments made via email replies are discarded by GitHub.

### Command line

Please provide the output from the `top` command in a comment here:

```text
top -l 3
```

pidloop, 2022-09-22:
[IDE-top-l-3.txt](https://github.com/arduino/arduino-ide/files/9622464/IDE-top-l-3.txt)

per1234, 2022-09-23:
>[IDE-top-l-3.txt](https://github.com/arduino/arduino-ide/files/9622464/IDE-top-l-3.txt)

I see there is one process using all the CPU:
```text
PID    COMMAND          %CPU TIME     #TH   #WQ #PORTS MEM    PURG   CMPRS  PGRP  PPID  STATE    BOOSTS      %CPU_ME %CPU_OTHRS UID FAULTS     COW     MSGSENT    MSGRECV    SYSBSD      SYSMACH     CSW        PAGEINS  IDLEW       POWER INSTRS      CYCLES     USER                   #MREGS RPRVT VPRVT VSIZE KPRVT KSHRD
56405  Arduino IDE Help 99.5 02:16.11 22/1  1   218    555M+  0B     0B     56393 56393 running  *0[1]       0.00000 0.00000    503 353556+    1817    46032+     30535+     115208+     182088+     155122+    0        1005+       99.9  10588127980 4567264198 ecdowney               N/A    N/A   N/A   N/A   N/A   N/A  
```

I would also like to get the full command line of this specific process from you. You can get that by running the `top` command again to determine the PID of the process, shown under the "PID" column of the table (it was `56405` in this case), then running a `ps` command for that PID:

```text
ps -p <PID>
```

(e.g., `ps -p 56405`)

Please post the output of that command here.

pidloop, 2022-09-23:
output of ps axuwww | grep 8579

`ecdowney          8579 100.6  2.7 74205524 895740   ??  R    12:47PM   1:53.47 /Applications/Arduino IDE 3.app/Contents/Frameworks/Arduino IDE Helper (Renderer).app/Contents/MacOS/Arduino IDE Helper (Renderer) --type=renderer --user-data-dir=/Users/ecdowney/Library/Application Support/arduino-ide --app-path=/Applications/Arduino IDE 3.app/Contents/Resources/app --no-sandbox --no-zygote --field-trial-handle=1718379636,7398052868659905054,1965251737215176558,131072 --disable-features=SpareRendererForSitePerProcess --lang=en-US --num-raster-threads=4 --enable-zero-copy --enable-gpu-memory-buffer-compositor-resources --enable-main-frame-before-activation --renderer-client-id=4 --no-v8-untrusted-code-mitigations --shared-files
`