arduino/arduino-ide #1485: some functions and constants doesn't colored like Arduino IDE 1.x

Nch-MOSFET, 2022-09-22:
### Describe the problem

some functions and constants doesn't colored like Arduino IDE 1.x

[image]

`sizeof` and `OUTPUT` must be coloed as light blue

the below image shows a same code in Arduino IDE 1.8.13

[image]

### To reproduce

I used this code

```
uint8_t pins[] = {1, 2, 3, 4, 5};

void setup() {
  for(uint8_t i = 0, i < sizeof(pins); i++) {
    pinMode(pins[i], OUTPUT);
  }
}

void loop() {}
```

### Expected behavior

The display of functions and constants should be the same as in the previous IDE

### Arduino IDE version

2.0.0

### Operating system

Windows

### Operating system version

Windows 10 Pro 21H1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-22:
Hi @Nch-MOSFET . Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/380.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-09-22 by per1234

Nch-MOSFET, 2022-09-22:
Sorry, I couldn't find it