arduino/arduino-ide #1491: Cores bundled with IDE installer or Support for Offline Installation

mlksahib, 2022-09-24:
### Describe the request

I love the new IDE 2.0 but can't use it in my remote work location where internet connection is too slow to be used.
Please add support for adding Arduino AVR Core without internet connection or bundle these basic cores with IDE installer.
Just as new libraries can be added using .zip files, new cores should also added in similar fashion i.e. by adding .zip files.

It would be great if you add arduino avr cores, and Atmega cores (like miniCore and mightyCore for programming bare micro-controllers with IDE) with IDE installer, so these can be installed on offline PCs 

### Describe the current behavior

New IDE 2.0 cant even start with out internet connection, which makes it unusable for some people who either don't have internet connection to their work PC or the internet connection is too slow to be used. IDE 1.x.x was great in this regard as it comes with Arduino-AVR core bundled with software installer. I also installed SAM-core for programming Arduino Due on one of my friends PC who didn't have a internet connection. Plus i installed miniCore for programming Atmega328P based board via USBASP.

### Arduino IDE version

IDE 2.0

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

koendv, 2022-10-02:
If internet or processor are slow, I use arduino-cli, e.g.:
```
arduino-cli lib install TaskScheduler
```
HTH

eptercom, 2023-06-30:
Simple issues like this have me downgrading to 1.8 and using other IDEs...