arduino/arduino-ide #1495: Changing font in Serial Monitor window (Mac OS)

janhesmert, 2022-09-25:
On Mac OS 10.14, the font of the Serial Monitor window cannot be changed via editing ~/Library⁩/⁨Arduino15⁩/preferences.txt. It should be possible to edit the chosen font and also the size via

serial.font=[Font name],[weight],[size]

according to the way it can be edited for the Editor window. However, this seems to have no effect.

oobed258, 2023-10-23:
Open the Arduino IDE.
Click on "File" in the top-left corner and select "Preferences."
In the "Preferences" window, you'll find a field labeled "Editor Font." Click on the "Change" button next to it.
This will open a font selection dialog where you can choose a new font, font size, and style (e.g., bold or italic). Select the font and settings you prefer and click "OK."

per1234, 2023-10-23:
@oobed258 I'm not sure which version or variant of Arduino IDE you are using, but Arduino IDE 2.x does not have the preference as you claim:

[image]

It is possible to configure the font used in the Arduino IDE editor view via the `editor.fontFamily` [advanced setting](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings) and the font used in the terminal via `terminal.integrated.fontFamily`. However, neither of those affect the font used by Serial Monitor.

oobed258, 2023-10-23:
sorry, i thought you were talking about font size

Zipgit, 2024-12-28:
Five years now....and this request still has not been addressed? I am surprised. 
I am using v2.3.4 of the IDE, on a Linux platform. 
I am really pleased that I can change the font of the editor. Intuitively, I would think that the Serial Monitor font would follow those editor-font changes (as the "default"), but that is not the behaviour we see right now.
As the font preference for the Serial Monitor does not automatically default to the Editor field's font, it would make sense to have the option to do that.  I find that the two typefaces side-by-side form a bit of a distraction.