arduino/arduino-ide #1500: Unable to connect to Arduino Cloud in Arduino IDE

LeonardoMazzoli, 2022-09-26:
### Describe the problem

While trying to connect to Arduino Cloud, after successful login in Arduino Cloud, the callback from the browser to the Arduino IDE doesn't work, it says 'localhost refuses to connect'. Tried with Safari, Chrome and Firefox, same problem

### To reproduce

Just try to connect to Arduino IDE

### Expected behavior

Successful linking between Arduino Cloud and the IDE

### Arduino IDE version

2.0

### Operating system

macOS

### Operating system version

Monterey

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-26:
Thanks for your report @LeonardoMazzoli. I'm going to ask you to run the Arduino IDE from the command line, then share the output. It may be that the logs printed there will provide some additional information that will assist in the investigation.

Please do this:

1. Open your **Applications** folder.
1. Open **Utilities**.
1. Double-click on "**Terminal**".
1. Type the following command:
   ```text
   "/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE"
   ```
1. Press **<kbd>Enter</kbd>**
1. Wait for the Arduino IDE to finish starting.
1. Attempt to connect to Arduino Cloud, just as you did before.
1. Switch back to the terminal window.
1. Press **<kbd>⌘</kbd>+<kbd>A</kbd>**.
   This will select all the text in the terminal window.
1. Press **<kbd>⌘</kbd>+<kbd>C</kbd>**.
   This will [copy](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the selected text to the clipboard.
1. Click on the "**Leave a comment**" field here on this GitHub issue.
1. Press **<kbd>⌘</kbd>+<kbd>V</kbd>**.
   This will [paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the output into the code block.
1. Click the <kbd>**Comment**</kbd> button to make the reply.

---

Please let me know if you have any questions or problems while following those instructions.

LeonardoMazzoli, 2022-09-26:
Hi @per1234. Thanks for your reply. Here you are:
```
Last login: Mon Sep 26 15:07:07 on ttys000
leonardomazzoli@Leonardos-MBP ~ % "/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE"
Starting backend process. PID: 9651
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///Users/leonardomazzoli/.theia'
root INFO Backend Object.initialize: 9.6 ms [Finished 2.447 s after backend start]
root INFO Backend Object.configure: 6.4 ms [Finished 2.447 s after backend start]
root INFO Backend MessagingContribution.onStart: 3.1 ms [Finished 2.447 s after backend start]
root INFO Backend OSBackendApplicationContribution.configure: 1.7 ms [Finished 2.523 s after backend start]
root INFO Backend DefaultWorkspaceServer.onStart: 1.2 ms [Finished 2.523 s after backend start]
root INFO Backend LocalizationBackendContribution.configure: 0.8 ms [Finished 2.524 s after backend start]
root WARN Backend LocalizationBackendContribution.initialize took longer than the expected maximum 50 milliseconds: 77.7 ms [Finished 2.526 s after backend start]
root INFO Backend TaskBackendApplicationContribution.onStart: 1.9 ms [Finished 2.526 s after backend start]
root INFO Backend ElectronTokenBackendContribution.configure: 4.6 ms [Finished 2.531 s after backend start]
root INFO Backend PluginDeployerContribution.initialize: 1.0 ms [Finished 2.531 s after backend start]
root INFO Backend ConfigServiceImpl.onStart: 5.7 ms [Finished 2.536 s after backend start]
root INFO Backend WebviewBackendSecurityWarnings.initialize: 3.8 ms [Finished 2.544 s after backend start]
root INFO Backend ArduinoDaemonImpl.onStart: 5.4 ms [Finished 2.546 s after backend start]
root INFO Backend NodeFileUploadService.configure: 10.1 ms [Finished 2.547 s after backend start]
daemon INFO Starting daemon from /Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
discovery-log INFO start
discovery-log INFO start new deferred
root INFO Backend BoardDiscovery.onStart: 2.4 ms [Finished 2.550 s after backend start]
root INFO Backend FileDownloadEndpoint.configure: 1.2 ms [Finished 2.551 s after backend start]
root INFO Backend AuthenticationServiceImpl.onStart: 0.9 ms [Finished 2.553 s after backend start]
root INFO Backend PluginApiContribution.configure: 1.8 ms [Finished 2.554 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.4 ms [Finished 2.556 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.3 ms [Finished 2.557 s after backend start]
root INFO Backend PlotterBackendContribution.configure: 3.1 ms [Finished 2.561 s after backend start]
root INFO Theia app listening on http://localhost:50100.
root INFO Finished starting backend application: 12.1 ms [Finished 2.568 s after backend start]
root WARN The local plugin referenced by local-dir:/Users/leonardomazzoli/.arduinoProIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/Users/leonardomazzoli/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/Users/leonardomazzoli/.arduinoIDE/extensions does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.5" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-cs" to a VS Code extension "vscode-language-pack-cs@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-es" to a VS Code extension "vscode-language-pack-es@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-it" to a VS Code extension "vscode-language-pack-it@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-ja" to a VS Code extension "vscode-language-pack-ja@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-nl" to a VS Code extension "vscode-language-pack-nl@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
[... cut ...]
discovery-log WARN Port '/dev/cu.Bluetooth-Incoming-Port Serial Port' was already available. Known boards before override: []
root INFO Attached boards and available ports changed:
root INFO No changes.
root INFO ------------------------------------------
root INFO server listening on http://localhost:9876
root ERROR Request login failed with error: Login timeout.
```

This part:
```
root INFO server listening on http://localhost:9876
root ERROR Request login failed with error: Login timeout.
```
came up after trying to login

kittaakos, 2022-10-05:
I am unsure if my issue is the same as this one, but login does not work anymore after clicking once on the `Sing In` button. Hence, users have a single chance per browser window to log in.

Steps to reproduce:
 - Open IDE2,
 - If you're logged in to Arduino Create, log out.
 - Click on the `Sign In` button. It opens the SSO login page in the default browser,
 - Ignore the SSO browser tab, and go back to IDE2,
 - Click on the `Sign In` button again,
 - Nothing happens.

Error on the backend:
```
root ERROR Uncaught Exception:  Error: Error listening to server
root ERROR Error: Error listening to server
    at Server.<anonymous> (/Users/a.kitta/dev/git/arduino-ide/arduino-ide-extension/lib/node/auth/authentication-server.js:52:20)
    at Server.emit (node:events:394:28)
    at emitErrorNT (node:net:1342:8)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
```

https://user-images.githubusercontent.com/1405703/194038126-8a626ae8-5d53-40a4-a16d-693f68f34fa1.mp4