arduino/arduino-ide #1501: no acknowledgement of read only file when saving

wilamsworld, 2022-09-26:
### Describe the problem

If you load a read only .ino then try to save the IDE does not notify that the file is Read Only and it appears to save to this file although it has not.

### To reproduce

load read only file
edit 
save

### Expected behavior

I would expect a popup or similar to notify the file was READ ONLY

### Arduino IDE version

2.0

### Operating system

Windows

### Operating system version

10

### Additional context

no additional context

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

thijstriemstra, 2022-09-29:
I'm amazed this is marked as "imperfection" when it can lead to data loss. Wow.

jl-esp, 2023-07-18:
Issues still exists in :
Version: 2.1.1
Date: 2023-06-30T16:04:40.277Z
CLI Version: 0.32.3

I get no notification that the file is read-only and unable to be saved.

kittaakos, 2023-07-18:
> I looked at the changelog for the nightly build and it listed:” - fix: warn user when IDE cannot save the sketch [[8f8b46f](https://github.com/arduino/arduino-ide/commit/8f8b46fa6fa051777a83c842964de49bf81f6196)]”
> I downloaded it and see the following:

👆 that was a fix for a different problem.

> Issues still exists in :
> Version: 2.1.1

It's expected. This issue is open as it is not implemented yet. The fix is available neither from the nightly nor from the release build, as it (kind of) depends on https://github.com/eclipse-theia/theia/pull/12354. Thanks for your patience.

jl-esp, 2023-07-18:
Thank you - at least this is an improvement from zero notification that I had before.

Anticipating the real fix.  I read the bug you posted and it looks promising.  A pop-up would be ideal.
 
Thanks!!