arduino/arduino-ide #1503: Tooltip too squashed

Edivad99, 2022-09-27:
### Describe the problem

The pop-up that opens shows the information that is too squashed and is difficult to read as well

### To reproduce

Place the mouse over a function

### Expected behavior

Better visibility
[image]

### Arduino IDE version

2.0.0

### Operating system

macOS

### Operating system version

macOS 12.6

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-27:
Thanks for your report @Edivad99. It is not clear to me what the problem is. Please provide more details.

>The pop-up that opens shows the information that is too squashed

Please provide a detailed description of what you mean by "squashed".

Are there specific parts of the tooltip that seem "squashed" to you? If so, which parts?

Please describe some specific changes that would cause it to seem less "squashed".

>is difficult to read as well

Please describe the exact problems that cause it to be difficult to read.

Are there specific parts of the tooltip that are difficult to read? If so, which parts?

Please describe some specific changes that would cause it to be less difficult to read.

Edivad99, 2022-09-27:
By squashed I mean that there is little space between the writing and that the words touch the gray lines.
- In "class String" the writing touches the gray line
- In "// In Stram" the writing is too attached to the line
- The word "instance-method readString" takes up a lot of space
- The fourth rectangle is a duplication of the information I have in the third rectangle

per1234, 2022-09-27:
Thanks!

>- In "class String" the writing touches the gray line
>- In "// In Stream" the writing is too attached to the line

I am also a little bothered by this. I mentioned it in passing in https://github.com/arduino/arduino-ide/issues/21

>- The word "instance-method readString" takes up a lot of space

This is tracked here: https://github.com/arduino/arduino-ide/issues/21

>The fourth rectangle is a duplication of the information I have in the third rectangle

Which board do you have selected in the **Tools > Board** menu in the Arduino IDE? I don't have this part when I did a quick check with the Uno selected:

[image]

---

In the end, I think the comment I made here will apply to both this and https://github.com/arduino/arduino-ide/issues/21:

https://github.com/arduino/arduino-ide/issues/21#issuecomment-949515760

This formatting is coming from the [**clangd** C++ language server](https://clangd.llvm.org/) that provides this information to the Arduino IDE, and perhaps some from the [**Eclipse Theia** IDE framework](https://theia-ide.org/) the IDE is built on or even the [**VS Code**](https://code.visualstudio.com/) application Theia takes some of its design from.

None of those projects are maintained by Arduino and I'm not sure whether anything can be done to improve on it from inside the Arduino IDE codebase hosted in this repo.

Edivad99, 2022-09-27:
If you pass the word readString over it and press command, the fourth rectangle appears

commit 5444395f (2022-10-07): Better tooltips.

fixes #1503
closed 2022-10-07 by per1234