arduino/arduino-ide #1506: Sketch > Export compiled Binary

Jens869, 2022-09-29:
### Describe the problem

Version: 2.0.0
Datum: 2022-09-14T07:08:28.861Z
CLI Version: 0.27.1 [a900cfb2]

Sketch > Export compiled Binary

Mac OS

the bin file is found in: /private/var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino-sketch-352F5EAEA50A9FC581E2424D67B8F822/VFD-8266-Uhr.ino.bin
but not in: /Users/-username-/Documents/Arduino/VFD-8266-Uhr/VFD-8266-Uhr.ino.bin

the subfolder is hard to reach

But it old Arduino IDE 1.x puts it in the right folder !

### To reproduce

install Arduino IDE 2.0 on Mac
load an example
press the keys: alt - cmd - s

### Expected behavior

place bin file in sketch folder

### Arduino IDE version

2.0.0

### Operating system

macOS

### Operating system version

MacOS 11.6.8 Intel

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-29:
The reason it is saved to that temporary folder is because the Arduino IDE 2.x opens example code in a temporary new sketch (as discussed at https://github.com/arduino/arduino-ide/issues/925). The IDE does this because example sketches are intended to be treated as "read-only" so that the user does not unintentionally make modifications to what is expected to be "known good" reference code provided by the library author and Arduino.

So I'm not convinced that Arduino IDE 1.x saving the compiled binary inside what is intended to be somewhat of a "read-only" location was actually "right". On the other hand, doing so shouldn't cause problems like allowing a modification to the example code would because the presence of the exported binary in a sketch folder shouldn't affect the IDE in any way.

The workaround for the problem of the example sketch being in an odd temporary folder is to use **File > Save As...** to save the temporary new sketch to a persistent location on your hard drive.

>the subfolder is hard to reach

Even in the temporary folder, it is sill just as easy to reach by selecting **Sketch > Show Sketch Folder** from the Arduino IDE menus.

kittaakos, 2022-09-29:
> Arduino IDE 1.x saving the compiled binary inside what is intended to be somewhat of a "read-only" location was actually "right".

IDE2 can prompt a dialog and guide users to save the temp sketch first when doing `Sketch` > `Export Compiled Binary`.

IDE 1.x shows this dialog when trying to save an example:

[image]

Jens869, 2022-09-29:
I'm sorry.
Arduino 1.x places the bin file in /Documents/Arduino/name/name.bin

Arduino 2.0 in /Documents/Arduino/name/build/esp8266-something/name.bin

It's also a problem that modern operating systems hide the path, but when I **upload the file via browser** then I have to **specify the entire path.**

There it is useless to have a shortened path in Finder/Explorer.

Unless you're using the terminal/shell/cmd.exe

per1234, 2022-09-30:
>Arduino 2.0 in /Documents/Arduino/name/build/esp8266-something/name.bin

This is intentional.

>It's also a problem that modern operating systems hide the path

Then change the operating system settings to have the right behavior. If that setting is not available, take it up with Apple.

> when I upload the file via browser then I have to I have to specify the entire path.

If that is a problem, you can move the file to a more convenient location.

closed 2022-09-30 by per1234

hemangjoshi37a, 2024-06-13:
where is this located in ubuntu ?