arduino/arduino-ide #1509: Minor bug: Special characters (in comments) not handled correct

CptHolzschnauz, 2022-10-01:
### Describe the problem

If there is a special character like ü, the IDE / Cloud combination (pull/push) forms this ü into Ã¶.

### To reproduce

Put a ü in a comment 

### Expected behavior

ü instead of Ã¶

### Arduino IDE version

2.0.0

### Operating system

macOS

### Operating system version

Monterey

### Additional context

I'm not 100 % sure but even if you corrected to ü in the sketch it's changing back after a while

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-01:
Hi @CptHolzschnauz. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/449.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-10-01 by per1234

CptHolzschnauz, 2022-10-01:
Hi per1234! Sure, sorry i didn't find #449 before. Have a good one!