arduino/arduino-ide #1510: Problems with ArduinoOTA.

Jens869, 2022-10-01:
### Describe the problem

Wemos D1 Mini ESP8266

1. using ArduinoOTA Example -> ArduinoOTA -> BasicOTA
edit  WIFI and password 
set OTA password (does it not work without password, small mistake.)
2. upload via USB-serial
3. connecting ESP8266 to USB power adapter
4. select IP as network port
5. upload via  network port

successful but Arduino IDE stays on upload.
To enable the next upload, ArduinoIDE must be closed and start again, otherwise the upload is no longer available. !!

### To reproduce

.

### Expected behavior

Arduino IDE should time out 
or finish the upload itself
or a stop button should be available

### Arduino IDE version

Version: 2.0.0 Datum: 2022-09-14T07:08:28.861Z CLI Version: 0.27.1 [a900cfb2]  Copyright © 2022 Arduino SA

### Operating system

macOS

### Operating system version

MacOS 11.7 Intel

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

Jens869, 2022-10-01:
Same behavior with:

Version: 2.0.1-nightly-20221001
Datum: 2022-10-01T03:58:39.348Z
CLI Version: 0.27.1 [a900cfb2]

Copyright © 2022 Arduino SA

per1234, 2022-10-01:
Hi @Jens869. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/1356.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-10-01 by per1234

Jens869, 2022-10-01:
I just watched this video
https://www.youtube.com/watch?v=iMPBpUdL-rk
There it is said that the upload function does not come back.
16:55
"the reason i had to create a thread is once you call this you are not going to ever get back to the draw function. and if you never get back to the draw function then how am i going the status in real time?" he said ...

Version: 2.0.1-nightly-20221001
Datum: 2022-10-01T03:58:39.348Z
CLI Version: 0.27.1 [a900cfb2]

Mac OS 11.7