arduino/arduino-ide #1515: Top part of UI disappears when opening recent sketch

aliphys, 2022-10-03:
### Describe the problem

When opening a recent sketch, the top part of the UI disappears.
[image]

### To reproduce

1. Open a recently opened sketch
2. Click on Output
3. Top part of window disappears

See video recording here: https://www.loom.com/share/f3e55c19d1434a128695cc5a9f7820d8

### Expected behavior

I would expect that the UI elements don't change, when I click on Output. 

### Arduino IDE version

20221003

### Operating system

Windows

### Operating system version

Windows 10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

r3inbowari, 2022-10-03:
Hi! I can't reproduce, but I noticed that the occluded problem seems to reappear when the window is minimized.
OS: Windows 11(21H2)

[image]

per1234, 2022-10-03:
Thanks @aliphys. Please open **File > Preferences** from the Arduino IDE menus and tell me whether the "**Automatic**" checkbox is checked or unchecked next to the "**Interface Scale**" preference, and what is the % value of that preference.

I encountered this bug some time back: https://github.com/arduino/arduino-ide/issues/779. I marked it as solved, but now I see it was solved only when using the default 100% interface scale. I can still reproduce the bug (both with the IDE version from the time of [the "fix"](https://github.com/arduino/arduino-ide/pull/1151), and with the latest nightly) when I use a larger interface scale value.

>I noticed that the occluded problem seems to reappear when the window is minimized.

I noticed this also. This was previously reported at https://github.com/arduino/arduino-ide/issues/662 and considered fixed by the same PR https://github.com/arduino/arduino-ide/pull/1151. As with https://github.com/arduino/arduino-ide/issues/779, I see the "fix" was only a fix when using the default interface scale. However, unlike https://github.com/arduino/arduino-ide/pull/1151 there was almost immediately a regression that caused the fix to be lost even with the default interface scale setting! I didn't do a full bisection, but I find the status bar is clipped slightly starting from 99664ee and seriously clipped as shown in your screenshot starting from 5ff9ce0.

aliphys, 2022-10-03:
@per1234 This is the screenshot for the **File > Preferences** menu
[image]
The **"Automatic"** checkbox for the **"Interface Scale"** is checked, and the value is set to 120%.

I tried seeing what happens when I resize the window (with the above settings) and can confirm your observations about https://github.com/arduino/arduino-ide/issues/779 . The top part only truncates when the window is smaller than a critical value.

Additionally, when the **"Automatic"** check box is disabled and the **"Interface Scale"** value is set to 100%, the truncation issues is resolved.

I have demonstrated both here: https://www.loom.com/share/c896d29dc69e4c39aec1b61ae09636ef

closed 2022-10-07 by per1234