arduino/arduino-ide #1524: No porting across of "Additional Boards Manager URL's" in preferences from previous Arduino version

Gerriko, 2022-10-04:
### Describe the problem

I have version 1.8.16 installed on my computer, which includes a number of URLs listed in the preferences "Additional Boards Manager URL's" section.

When I installed version 2.0, it correctly lists all those boards as installed under the Tools>Board menu option. However the URL's are missing under preferences.

### To reproduce

Have a previous version of Arduino installed.
Have some unique boards URL references listed under preferences "Additional Boards Manager URL's" section
Install version 2.0
Check that the boards are found under the Board menu option in version 2.
Now open up the preference dialog box.
Note that those URL's for those specific boards are not listed here.

### Expected behavior

The URL's listed in a previous version need to be listed in version 2.0 preferences "Additional Boards Manager URL's" section.

### Arduino IDE version

2.0

### Operating system

Linux

### Operating system version

20.04

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-04:
Thanks for your report @Gerriko. I agree this is one of the most problematic aspects of the Arduino IDE 1.x to 2.x migration.

The installed boards platforms are recognized just fine by the IDE, so there isn't an immediate impact after migration. The problem comes later as the user's 3rd party boards platforms become increasingly outdated as they are unaware of the availability of updates.

The Arduino Tooling Team discussed this and the chosen solution was to make the IDE notify the user when it detects the presence of a boards platform installed via Boards Manager without an associated package index URL in the "**Additional Boards Manager URLs**" preference. That requires work to be done first in Arduino CLI, so the proposal is tracked in that repository:

https://github.com/arduino/arduino-cli/issues/1768

The more obvious solution would be to make Arduino IDE 2.x read the existing preferences from the Arduino IDE 1.x preferences file (the two versions use completely different preferences stores). However, that code is only of value during this transition period. The alternative proposal has ongoing value even for those who never used Arduino IDE 1.x. The reason is a usage pattern I have observed where users will later remove URLs for installed platforms from their "**Additional Boards Manager URLs**" preference. They do this for either of two reasons:

- They don't realize the field is a list that supports multiple URLs, so they replace the previous URL when adding a new URL to install another platform
- They do a "tidy up" of the preferences

The users generally think of this URL only in the context of installing a boards platform, not realizing that it also provides the important update capability. So they don't understand the impact of the removal.

That pattern becomes more likely with the more resilient design of Arduino IDE 2.x, where the platform is recognized even without a URL (in Arduino IDE 1.x, the boards disappeared, which is a poor user experience, but also forced the replacement of the URL). The proposed feature makes sure the problem of the missing will be clearly communicated to the user regardless of why it went missing.

bobcmartin, 2022-10-19:
Hi
I have also noticed that updates to additional cores save under packages are not properly updated in 2.o but are in 1.8.x.  For example I am reformatting some entries in programmers.txt contained in the DxCore from Spence Konde and note that changes are reflected in the Arduino 1.8.x IDE but are ignored in the 2.0 IDE.  I am restarting both 1.8.x and 2.0 to ensure cached information is cleared but does 2.0 maintain a separate database also shared by the CLI ?
IDE 2.0 seemed to just borrow the externa; board / core support it seems there's some other database being used