arduino/arduino-ide #1527: Arduino IDE 2.0 hangs on startup in GUI mode - MacOS Monterey.

erongd, 2022-10-05:
### Describe the problem

 I have installed the Arduino IDE 2.0 on MacOS Monterey (12.6). It is a late-2019 16" macbook pro. When start the IDE in UI mode through spotlight or application folder, the IDE hangs. No splash screen, only the icon jumping in the dock eventually stuck with "Application not responding". However when I start the IDE in command line mode through terminal, the IDE starts just fine. Due to hang state I could not get any logs. It is very strange behavior.

I start to experience the issue in rc-9, thought it is due to beta. But seems the 2.0 GA version has the same problem. Once the IDE starts from the command line everything seems to work just fine.

### To reproduce

Don't know what would be needed to reproduce it.

### Expected behavior

Arduino IDE should start normally with splash screen then the actual IDE open on desktop.

### Arduino IDE version

2.0.1-nightly-20221004

### Operating system

macOS

### Operating system version

12.6

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

CptHolzschnauz, 2022-10-05:
Can't confirm. I installed it on a 2013 MacPro, on a iMac27 and on a M1 Air (All 12.6) without any problems or the issue you described here.

per1234, 2022-10-05:
Hi @erongd. Thanks for you report.

The IDE automatically writes log files while it is running. I think these logs might contain some information that will allow us to identify the problem so I'm going to ask you to provide the log file from your computer.

Please do this:

1. Attempt to start the Arduino IDE "in UI mode", as you did before.
1. Wait some time to be sure the application is in the hung state.
1. Open the issue page here on GitHub:
   https://github.com/arduino/arduino-ide/issues/1527#issue-comment-box
1. Click the "**Attach files by dragging & dropping, selecting or pasting them**" at the bottom of the "**Leave a comment**" field.
1. Select the latest `.log` file from this folder (the files are named according to the date):
   ```text
   ~/Library/Logs/Arduino IDE/
   ```
1. Click the <kbd>**Open**</kbd> button to attach the file to the comment.
1. You will now see an "**Uploading your files... (1/1)**" status indicator at the bottom of the "**Leave a comment**" field. Wait until that disappears.
1. Click the **<kbd>Comment</kbd>** button to publish the comment.

---

Please let me know if you have any questions or problems while following those instructions.

erongd, 2022-10-06:
Unfortunately start from UI mode does not produce any log before it hung.

Arcanum-XIII, 2022-10-24:
On my Mac, (Montery 12.5.1, Air M1), I got:

`root ERROR Failed to start the backend application:
root ERROR Error: listen EADDRNOTAVAIL: address not available 127.0.0.1
    at Server.setupListenHandle [as _listen2] (node:net:1298:21)
    at listenInCluster (node:net:1363:12)
    at GetAddrInfoReqWrap.doListen [as callback] (node:net:1501:7)
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (node:dns:73:8)
root ERROR Uncaught Exception:
root ERROR Error: listen EADDRNOTAVAIL: address not available 127.0.0.1
    at Server.setupListenHandle [as _listen2] (node:net:1298:21)
    at listenInCluster (node:net:1363:12)
    at GetAddrInfoReqWrap.doListen [as callback] (node:net:1501:7)
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (node:dns:73:8)`

Then some error regarding missing plugins in my localUser .arduinoIDE config. It's currently empty so...

But I guess the Backend without connexion would explain most of the issue.

dave0mi, 2023-01-19:
Problem: The IDE splash screen appears and hangs forever (2.0.3 IDE for Mac).

This is the startup log in MacOS Monterey 12.1. It is an M1, not sure if that is relevant because, java?
(Sorry for pasting the log inline, but my interface resisted every effort to upload the file.)

2023-01-19 12:24:49 Using browser-only version of superagent in non-browser environment
2023-01-19 12:24:49 Configuration directory URI: 'file:///Users/dilstrup/.theia'
2023-01-19 12:24:49 Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2023-01-19 12:24:49 2023-01-19T20:24:49.626Z root INFO Backend Object.initialize: 2.1 ms [Finished 0.476 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.626Z root INFO Backend Object.configure: 1.6 ms [Finished 0.476 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.626Z root INFO Backend MessagingContribution.onStart: 0.8 ms [Finished 0.476 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.626Z root INFO Backend OSBackendApplicationContribution.configure: 0.2 ms [Finished 0.494 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.627Z root INFO Backend DefaultWorkspaceServer.onStart: 0.2 ms [Finished 0.495 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.627Z root INFO Backend PluginLocalizationBackendContribution.configure: 0.1 ms [Finished 0.495 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.628Z root INFO Backend PluginLocalizationBackendContribution.initialize: 18.1 ms [Finished 0.495 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.628Z root INFO Backend TaskBackendApplicationContribution.onStart: 0.1 ms [Finished 0.495 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.628Z root INFO Backend ElectronTokenBackendContribution.configure: 0.4 ms [Finished 0.495 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.628Z root INFO Backend PluginDeployerContribution.initialize: 0.2 ms [Finished 0.495 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.629Z root INFO Backend ConfigServiceImpl.onStart: 0.4 ms [Finished 0.496 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.629Z root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.2 ms [Finished 0.496 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.629Z root INFO Backend ArduinoDaemonImpl.onStart: 0.6 ms [Finished 0.497 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.629Z root INFO Backend NodeFileUploadService.configure: 0.9 ms [Finished 0.497 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.629Z daemon INFO Starting daemon from /Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
2023-01-19 12:24:49 2023-01-19T20:24:49.629Z discovery-log INFO start
2023-01-19 12:24:49 2023-01-19T20:24:49.629Z discovery-log INFO start new deferred
2023-01-19 12:24:49 2023-01-19T20:24:49.629Z root INFO Backend HostedPluginLocalizationService.initialize: 0.3 ms [Finished 0.497 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.629Z root INFO Backend BoardDiscovery.onStart: 0.4 ms [Finished 0.497 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.629Z root INFO Backend FileDownloadEndpoint.configure: 0.3 ms [Finished 0.497 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.629Z root INFO Backend AuthenticationServiceImpl.onStart: 0.2 ms [Finished 0.498 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.629Z root INFO Backend PluginApiContribution.configure: 0.2 ms [Finished 0.498 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.630Z root INFO Backend HostedPluginReader.configure: 0.1 ms [Finished 0.498 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.630Z root INFO Backend HostedPluginReader.configure: 0.0 ms [Finished 0.498 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.630Z root INFO Backend PlotterBackendContribution.configure: 0.2 ms [Finished 0.499 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.631Z root INFO Theia app listening on http://localhost:57465.
2023-01-19 12:24:49 2023-01-19T20:24:49.631Z root INFO Finished starting backend application: 1.7 ms [Finished 0.500 s after backend start]
2023-01-19 12:24:49 2023-01-19T20:24:49.634Z root WARN The local plugin referenced by local-dir:/Users/dilstrup/.arduinoIDE/plugins does not exist.
2023-01-19 12:24:49 2023-01-19T20:24:49.634Z root WARN The local plugin referenced by local-dir:/Users/dilstrup/.arduinoIDE/extensions does not exist.
2023-01-19 12:24:49 2023-01-19T20:24:49.634Z root WARN The local plugin referenced by local-dir:/Users/dilstrup/.arduinoProIDE/plugins does not exist.
2023-01-19 12:24:49 2023-01-19T20:24:49.646Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-01-19 12:24:49 2023-01-19T20:24:49.646Z root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@1.5.1" with engines:
2023-01-19 12:24:49 2023-01-19T20:24:49.646Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-01-19 12:24:49 2023-01-19T20:24:49.646Z root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.5" with engines:
2023-01-19 12:24:49 2023-01-19T20:24:49.646Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-01-19 12:24:49 2023-01-19T20:24:49.646Z root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
[... cut ...]
2023-01-19 12:24:50 2023-01-19T20:24:50.118Z root INFO Deployed backend plugin "ms-ceintl.vscode-language-pack-zh-hans@1.53.2" from "/Applications/Arduino IDE.app/Contents/Resources/app/plugins/vscode-language-pack-zh-hans/extension": 19.3 ms [Finished 0.984 s after backend start]
2023-01-19 12:24:50 2023-01-19T20:24:50.118Z root INFO Deploy plugins list: 22.1 ms [Finished 0.987 s after backend start]
2023-01-19 12:24:50 2023-01-19T20:24:50.720Z root WARN Frontend a.configure took longer than the expected maximum 100 milliseconds: 230.1 ms [Finished 0.852 s after frontend start]

dave0mi, 2023-01-19:
Follow up on the above posting with the extensive log ...
(Re: Problem: The IDE splash screen appears and hangs forever (2.0.3 IDE for MacOS Monterey 12.1). 

After simply quitting the application and trying again, the IDE is able to start successfully (on my system).
Is it possible that the initial start is simply timing out on some of the 'first-run' configuration?

kittaakos, 2023-01-20:
The problem on your side is unrelated to any timing or missing `localhost` (https://github.com/arduino/arduino-ide/issues/1708#issuecomment-1332623299):

```
2023-01-19 12:24:49 2023-01-19T20:24:49.740Z daemon INFO �[36mINFO�[0m[0000] Config file not found, using default values  
```

```
ERROR Error: Could not read the CLI configuration file.
    at CoreClientProvider.evaluateErrorStatus (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/core-client-provider.js:188:20)
    at ClientReadableStreamImpl.<anonymous> (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/core-client-provider.js:175:36)
    at ClientReadableStreamImpl.emit (node:events:394:28)
    at endReadableNT (node:internal/streams/readable:1331:12)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
```

Most likely, when IDE2 started the first time on your macOS, it asked for permission to put the config file into `~/.arduinoIDE/arduino-cli.yaml` location, and you rejected it. But this is a guess. Try to give the Arduino IDE.app read-write access if the issue persists:
`System Preferences` > `Security & Privacy` > `Privacy` > `Files and Folders` and enable both. On my macOS, it looks like this:

[image]

If you cannot fix the problem by changing the permissions, please open a new issue in this repo or ask on the Arduino Forum: https://forum.arduino.cc/c/software/arduino-ide-2-0/93.

> java?

There is no Java.

stefannilsson, 2025-01-03:
Had the same issues on Apple M1, Sequoia 15.0.1 (24A348):
```
Arduino IDE 2.3.5-nightly-20241212
Checking for frontend application configuration customizations. Module path: /Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/electron-main.js, destination 'package.json': /Applications/Arduino IDE.app/Contents/Resources/app/package.json
Setting 'theia.frontend.config.appVersion' application configuration value to: "2.3.5-nightly-20241212" (type of string)
Setting 'theia.frontend.config.cliVersion' application configuration value to: "1.1.1" (type of string)
Setting 'theia.frontend.config.buildDate' application configuration value to: "2024-12-12T03:23:28.481Z" (type of string)
Frontend application configuration after modifications: {"applicationName":"Arduino IDE","defaultTheme":{"light":"arduino-theme","dark":"arduino-theme-dark"},"defaultIconTheme":"none","electron":{"windowOptions":{}},"defaultLocale":"","validatePreferencesSchema":false,"preferences":{"window.title":"${rootName}${activeEditorShort}${appName}","files.autoSave":"afterDelay","editor.minimap.enabled":false,"editor.tabSize":2,"editor.scrollBeyondLastLine":false,"editor.quickSuggestions":{"other":false,"comments":false,"strings":false},"editor.maxTokenizationLineLength":500,"editor.bracketPairColorization.enabled":false,"breadcrumbs.enabled":false,"workbench.tree.renderIndentGuides":"none","explorer.compactFolders":false},"appVersion":"2.3.5-nightly-20241212","cliVersion":"1.1.1","buildDate":"2024-12-12T03:23:28.481Z"}
Starting backend process. PID: 6828
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///Users/user1/.arduinoIDE'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2025-01-03T17:17:07.686Z root INFO Backend Object.initialize: 39.7 ms [Finished 1.129 s after backend start]
2025-01-03T17:17:07.702Z root INFO Backend Object.configure: 27.2 ms [Finished 1.130 s after backend start]
2025-01-03T17:17:07.702Z root INFO Backend s.onStart: 12.2 ms [Finished 1.130 s after backend start]
2025-01-03T17:17:07.711Z root INFO Backend s.configure: 15.9 ms [Finished 1.210 s after backend start]
2025-01-03T17:17:07.729Z root INFO Backend d.onStart: 5.4 ms [Finished 1.210 s after backend start]
2025-01-03T17:17:07.732Z root INFO Backend d.configure: 3.4 ms [Finished 1.214 s after backend start]
2025-01-03T17:17:07.746Z root WARN Backend d.initialize took longer than the expected maximum 50 milliseconds: 84.8 ms [Finished 1.221 s after backend start]
2025-01-03T17:17:07.756Z root INFO Backend c.onStart: 7.5 ms [Finished 1.221 s after backend start]
2025-01-03T17:17:07.763Z root INFO Backend a.configure: 19.4 ms [Finished 1.241 s after backend start]
2025-01-03T17:17:07.767Z config INFO >>> Initializing CLI configuration...
2025-01-03T17:17:07.769Z root INFO Backend c.initialize: 6.8 ms [Finished 1.248 s after backend start]
2025-01-03T17:17:07.770Z root INFO Backend x.onStart: 17.1 ms [Finished 1.261 s after backend start]
2025-01-03T17:17:07.776Z config INFO Loading CLI configuration from /Users/user1/.arduinoIDE/arduino-cli.yaml...
2025-01-03T17:17:07.780Z daemon INFO Starting daemon from /Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/resources/arduino-cli...
2025-01-03T17:17:07.780Z root INFO Backend a.initialize: 0.4 ms [Finished 1.269 s after backend start]
2025-01-03T17:17:07.780Z root INFO Backend w.onStart: 14.2 ms [Finished 1.282 s after backend start]
2025-01-03T17:17:07.780Z root INFO Backend r.configure: 28.4 ms [Finished 1.292 s after backend start]
2025-01-03T17:17:07.781Z discovery-log INFO start
2025-01-03T17:17:07.781Z discovery-log INFO start new deferred
2025-01-03T17:17:07.781Z root INFO Backend d.initialize: 6.3 ms [Finished 1.300 s after backend start]
2025-01-03T17:17:07.781Z root INFO Backend y.onStart: 10.7 ms [Finished 1.305 s after backend start]
2025-01-03T17:17:07.781Z root INFO Backend r.configure: 5.8 ms [Finished 1.306 s after backend start]
2025-01-03T17:17:07.781Z root INFO Backend a.onStart: 3.6 ms [Finished 1.317 s after backend start]
2025-01-03T17:17:07.781Z root INFO Backend h.configure: 3.0 ms [Finished 1.317 s after backend start]
2025-01-03T17:17:07.781Z root INFO Backend s.configure: 0.5 ms [Finished 1.317 s after backend start]
2025-01-03T17:17:07.781Z root INFO Backend l.configure: 0.1 ms [Finished 1.317 s after backend start]
2025-01-03T17:17:07.809Z root INFO Theia app listening on http://127.0.0.1:49314.
2025-01-03T17:17:07.810Z root INFO Finished starting backend application: 27.5 ms [Finished 1.344 s after backend start]
2025-01-03T17:17:07.869Z root WARN The local plugin referenced by local-dir:/Users/user1/.arduinoIDE/plugins does not exist.
2025-01-03T17:17:07.869Z root WARN The local plugin referenced by local-dir:/Users/user1/.arduinoIDE/extensions does not exist.
2025-01-03T17:17:08.169Z daemon INFO time="2025-01-03T18:17:07+01:00" level=info msg="arduino-cli version 1.1.1"
time="2025-01-03T18:17:07+01:00" level=info msg="Using config file: /Users/user1/.arduinoIDE/arduino-cli.yaml"
time="2025-01-03T18:17:07+01:00" level=info msg="Executing `arduino-cli daemon`"
Daemon is now listening on 127.0.0.1:49315
{"IP":"127.0.0.1","Port":"49315"}
2025-01-03T17:17:08.169Z daemon INFO Daemon is running.
2025-01-03T17:17:08.186Z config INFO Loading CLI configuration from /Users/user1/.arduinoIDE/arduino-cli.yaml...
[... cut ...]
2025-01-03T17:17:16.116Z root WARN Frontend D.onStart took longer than the expected maximum 100 milliseconds: 476.5 ms [Finished 7.789 s after frontend start]
```

Tried the nightly build (20241212) but no success.

Removed `~/Documents/Arduino` and re-created this folder, and opened **Finder** and right clicked on **Keep Downloaded**. And now it's working!
Not sure if it was related to wiping the content of the folder, or if it's some async task like downloading all folder content (if offloaded to iCloud when using this) never finishes (or takes "forever")?