arduino/arduino-ide #1532: Option to start new sketch when IDE started

smmakerspace, 2022-10-05:
### Describe the request

Please add an option to start a blank sketch at startup.  

### Describe the current behavior

Currently, when staring up the Arduino IDE 2.0, the last sketch is automatically opened.  

### Arduino IDE version

2.0

### Operating system

Windows

### Operating system version

Win10

### Additional context

I teach in a computer lab with shared computers and students are always forgetting to open a new sketch and are overwriting the previous student's work.  Thank you!

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

Careless, 2023-05-10:
would be nice if this was integrated into the new Arduino IDE.

kind of obnoxious to have the last sketch opened up and show its files and I have to click 'new sketch' before I close this one.

give the user the option to open a new sketch on startup in the preferences. it's not that hard to give the user some choice in this matter.

KirAsh4, 2023-06-18:
This has always been an issue. With version 1, users had to manually edit the preferences file to disable this, Now in version 2 there's no way to do that at all. I don't understand why it's forced upon users.

nicknugget, 2023-09-05:
I agree with that. The user should be able to choose whether to start with the last edited sketch or an empty workbench. It's bad UI design to just force that on the user.