arduino/arduino-ide #1534: Linux Mint.  IDE auto-inserts spurious carriage returns, backspaces, and tabs

ceptimus, 2022-10-05:
### Describe the problem

When editing a sketch, cursor randomly jumps to start of next line.  Sometimes, indenting goes haywire - as though the IDE self-inserts a newline, followed by one or more backspaces.  I've not noticed it randomly deleting characters I've already typed, however.
Also, (noticable by selecting a block) there is lots of trailing whitespace, probably tabs, at the ends of lines.
No problems on the same PC using IDE 1.8.19, nor with other non-Arduino applications and text editors.

I've seen reports of this issue, but from a year or more ago, and they were closed.

### To reproduce

Try running the IDE under Linux Mint Cinnamon, current latest version.

### Expected behavior

You may see the IDE auto-inserting characters - though I suppose it may also only do it on some PCs.

### Arduino IDE version

2.0.0 Date 2022-09-14T07:05:28.855Z  CLI Version 0.27.1 [a900cfb2]

### Operating system

Linux

### Operating system version

Linux Mint 21 Cinnamon 5.4.12

### Additional context

Intel Core i7-7740X 16GB RAM, NVIDIA GeForce GTX 1080 graphics card

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-05:
Hi @ceptimus. Thanks for your report.

> When editing a sketch, cursor randomly jumps to start of next line

We have an previous issue we are using to track this bug: https://github.com/arduino/arduino-ide/issues/437

> I've not noticed it randomly deleting characters I've already typed, however.
Also, (noticable by selecting a block) there is lots of trailing whitespace, probably tabs, at the ends of lines.

I don't remember receiving a previous report of this, but since this bug is related to the IDE's handling of trailing whitespace, it might very well have the same cause.

Please try this and then comment here to let me know the results:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "**Command Palette**".
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
1. A "**Preferences**" tab will now open in the Arduino IDE. In the "**Search Settings**" field of that tab, type `editor.trimAutoWhitespace`
1. Uncheck the checkbox next to "**`☑` Remove trailing auto inserted whitespace.**"
1. Click the **X** icon on the "**Preferences**" tab.
1. Perform whatever editing actions you were doing previously that produced the unexpected trailing whitespace.

Please note that the workaround for https://github.com/arduino/arduino-ide/issues/437 I describe above disables the IDE's automatic removal of whitespace that was generated for indenting the new line to match the previous line when that whitespace would become trailing whitespace (see my explanation [here](https://github.com/arduino/arduino-ide/issues/437) for details). So it is expected for that trailing whitespace will now occur. But your description indicates there is other completely unexpected trailing whitespace. It is the effect on that whitespace I am interested in.

per1234, 2022-10-11:
Hi @ceptimus. Did you ever get a chance to try the experiment I requested in my previous reply? If so, what was the result?

ceptimus, 2022-10-11:
> Hi @ceptimus [1]. Did you ever get a chance to try the experiment I
> Links:

Hi, yes, sorry for not replying sooner.

When I made the alteration you suggested, the IDE becomes very usable, 
and I didn't notice any unexpected whitespace.  I just went back to 
original settings to try it again.

What happens is that when the cursor is on a blank line, but indented 
(which, of course, is usually the case with C++ code) is that the cursor 
spontaneously jumps back to column 1.  Sometimes it jumps back to the 
start of the current line, and sometimes to the start of the next line 
below.  Any tabs / spaces that were on the line, doing the indenting, 
remain "in front of" the cursor.  So then I then press the tab key to 
get the cursor back where I want it and type / edit the line.  There's 
now some extra whitespace beyond the cursor.  That whitespace doesn't 
normally get trimmed.  It stays there, invisible of course, unless I 
select a block of text, either by using the mouse, or by holding down 
the shift key while navigating with the cursor keys.

When the cursor jumps to the start of a line that already contains text, 
it's natural to use the cursor keys to get the cursor back where I want, 
rather than pressing the TAB or spacebar - so that doesn't introduce 
fresh whitespace.

I'm not really bothered about the trailing whitespace - and I could 
always use some command-line tool to strip it anyway, if it did bother 
me - but the cursor jumping around with a mind of its own makes the IDE 
unusable.

I'll go back to the settings you suggested and that will work fine for 
me till there's a 'proper fix' :)  If you want to see the cursor jumping 
in action, I can make and upload a YouTube video of it happening.  It's 
not just an occasional jump - it's about once every two or three 
seconds!

per1234, 2022-10-12:
Thanks for the update.

>I can make and upload a YouTube video of it happening.

I appreciate the offer. Someone was already kind enough to do this and it really was helpful for understanding the nature of the problem: https://github.com/arduino/arduino-ide/issues/476#issue-970790781

Since that time, I have also been affected by this issue on one of my computers, so I definitely am familiar with the behavior.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the original issue https://github.com/arduino/arduino-ide/issues/437.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-10-12 by per1234

ceptimus, 2022-10-12:
Here's my quick YouTube video, demonstrating the cursor-jump bug, under 
Linux Mint 21 Cinnamon.

https://youtu.be/ea7-Tt361nA

Thanks for your help.