arduino/arduino-ide #1536: Column indicator

thebigMoose, 2022-10-05:
### Describe the request

Would sure appreciate a settable column indicator (a faint vertical line) useful for alignment of comments.  Something like MPLAB X IDE has.

### Describe the current behavior

I do not believe this feature currently exists.

### Arduino IDE version

2.0.0

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-10-05:
Hi @thebigMoose. Thanks for your request. I'm happy to tell you the IDE already has this capability, which can be configured via [the advanced settings](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings). I'll provide instructions:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "**Command Palette**".
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
1. A "**Preferences**" tab will now open in the Arduino IDE. In the "**Search Settings**" field of that tab, type `editor.rulers`
1. You will now see an "**Editor: Rulers**" setting. Under the "Render vertical rulers after a certain number of monospace characters ..." description, there is an "**Edit in settings.json**" link. Click that link.
   ❗ [A bug](https://github.com/arduino/arduino-ide/issues/1537) makes the link very difficult to see if you are using the light theme, but it is there if you look carefully.
1. A file named `settings.json` will now open in an editor tab in the IDE. You will see a line that looks like this:
   ```json
   "editor.rulers": []
   ```
   Add the column number at which you want the ruler to be shown inside the square brackets. For example, if you wanted it at column 120, it should look like this:
   ```json
   "editor.rulers": [120]
   ```
   **ⓘ** You can create multiple rulers by separating the column numbers with commas.
1. Click the **X** icon on the "**settings.json**" tab.
1. Click the **X** icon on the "**Preferences**" tab.

It seem the setting does not take effect reliably in the currently open IDE window, but if you restart the IDE you should now see the ruler at the configured position.

closed 2022-10-05 by per1234

thebigMoose, 2022-10-05:
per1234 Thank You!  I would never have figured that out on my own.  Thanks for being kind to this ol' guy...  your instructions were perfect!