arduino/arduino-ide #1540: Uninstalling platform causes IDE to stop working

AlbyIanna, 2022-10-06:
### Describe the problem

_Sometimes_, if you have a selected board that belongs to a platform and you try to uninstall that platform, the IDE will stop to work. No useful feedback is given, the platform will appear to be installed and any action you perform that uses the backend (like compiling a sketch, creating a new file or updating a library) will fail. Quitting the IDE and reopening it will show that the platform has actually been uninstalled.

https://user-images.githubusercontent.com/6939054/194276310-893060e5-623d-4613-82bc-7dc110d433bf.mov

### To reproduce

I wasn't able to reproduce it consistently because sometimes it works just fine, but repeating the steps a few times would always lead to the reported error.

1. Open the Boards Manager from **Tools > Board > Boards Manager...**
2. Search and install the **Arduino MBED OS Nano Boards** platform (only if you don't have it already installed)
3. Select the **Arduino Nano RP2040 Connect** from **Tools > Board > Arduino MBED OS Nano Boards > Arduino Nano RP2040 Connect**
4. From the Boards Manager, **UNINSTALL** the **Arduino MBED OS Nano Boards**
5. 😢 Output in the console says boards are uninstalled, but the **Arduino MBED OS Nano Boards** in the Boards Manager still looks installed and the menu item **Arduino MBED OS Nano Boards** is still present in the **Tools > Board** menu. Also major IDE features will stop working.

### Expected behavior

1. Open the Boards Manager from **Tools > Board > Boards Manager...**
2. Search and install the **Arduino MBED OS Nano Boards** platform (only if you don't have it already installed)
3. Select the **Arduino Nano RP2040 Connect** from **Tools > Board > Arduino MBED OS Nano Boards > Arduino Nano RP2040 Connect**
4. From the Boards Manager, **UNINSTALL** the **Arduino MBED OS Nano Boards**
5. 😸 Output in the console says boards are uninstalled, and the IDE won't stop working

### Arduino IDE version

2.0.1-nightly-20221006

### Operating system

macOS

### Operating system version

12.3.1 (21E258)

### Additional context

This bug never happened to me when I had a selected board from a different platform.

Also, I want to make clear that this bug didn't happen consistently to me, sometimes I met the expected behaviour, sometime I didn't.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

AlbyIanna, 2022-10-06:
I don't think so. These the errors I get from the backend:
```
root ERROR Request uninstall failed with error: 14 UNAVAILABLE: Connection dropped Error: 14 UNAVAILABLE: Connection dropped
    at Object.callErrorFromStatus (/Users/albertoiannaccone/Workspace/arduino-ide/node_modules/@grpc/grpc-js/build/src/call.js:31:26)
    at Object.onReceiveStatus (/Users/albertoiannaccone/Workspace/arduino-ide/node_modules/@grpc/grpc-js/build/src/client.js:349:49)
    at Object.onReceiveStatus (/Users/albertoiannaccone/Workspace/arduino-ide/node_modules/@grpc/grpc-js/build/src/client-interceptors.js:328:181)
    at /Users/albertoiannaccone/Workspace/arduino-ide/node_modules/@grpc/grpc-js/build/src/call-stream.js:187:78
    at processTicksAndRejections (node:internal/process/task_queues:78:11)

discovery-log ERROR Unexpected error occurred during the boards discovery. Error: 14 UNAVAILABLE: No connection established
    at Object.callErrorFromStatus (/Users/albertoiannaccone/Workspace/arduino-ide/node_modules/@grpc/grpc-js/build/src/call.js:31:26)
    at Object.onReceiveStatus (/Users/albertoiannaccone/Workspace/arduino-ide/node_modules/@grpc/grpc-js/build/src/client.js:409:49)
    at Object.onReceiveStatus (/Users/albertoiannaccone/Workspace/arduino-ide/node_modules/@grpc/grpc-js/build/src/client-interceptors.js:328:181)
    at /Users/albertoiannaccone/Workspace/arduino-ide/node_modules/@grpc/grpc-js/build/src/call-stream.js:187:78
    at processTicksAndRejections (node:internal/process/task_queues:78:11)
```

It doesn't look the same as #1529, right?

kittaakos, 2022-10-06:
Are there any errors in the backend logs, or only the `Connection dropped` and `No connection established`?

per1234, 2022-10-11:
In case you want more information about the cause of an error from the [`@grpc/grpc-js` package](https://github.com/grpc/grpc-node), there are some environment variables that control the verbosity of the output:

https://github.com/grpc/grpc-node/blob/master/TROUBLESHOOTING.md