arduino/arduino-ide #1542: It‘s difficult to change the font size

r3inbowari, 2022-10-06:
### Describe the problem

The font size value is always locked at 8 or 72. 
Related to #1502.

### To reproduce

https://user-images.githubusercontent.com/30739857/194424678-63930371-0064-451d-b7e4-4d91a2f91e79.mp4

### Expected behavior

Adjustable size range via keyboard.

### Arduino IDE version

2.0.1-nightly-20221007

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

AlbyIanna, 2022-10-07:
Thank you for reporting this @r3inbowari. Unfortunately I cannot reproduce this on MacOS. Later I will give it a try on a Windows machine. In the meantime, it would be very helpful if you could describe what happens when you perform the following actions with the Preferences open:
- click on the arrows in the **Interface scale** input field; every time you click, the value should increase or decrease by 20%.
- with the **Automatic** flag **checked**, press CTRL + "+"; you should see all the IDE interface being zoomed in and the **Interface scale** input field should increase its value by 20%
- with the **Automatic** flag **unchecked**, press CTRL + "+"; you should see the font size in the code section increasing in size and the **Editor font size** input field should increase its value by 2

Do they work as expected?

r3inbowari, 2022-10-07:
> Thank you for reporting this @r3inbowari. Unfortunately I cannot reproduce this on MacOS. Later I will give it a try on a Windows machine. In the meantime, it would be very helpful if you could [...]

Yes, They work fine. 

Reproduce: For example, the current fontsize is 72 and we can not adjust to 16 only by keyboard.
This problem also exists on linux.

AlbyIanna, 2022-10-07:
> Reproduce: For example, the current fontsize is 72 and we can not adjust to 16 only by keyboard.

Oh sorry, initially I misinterpreted the issue. That is surely annoying! I've just opened a PR (#1547) with a possible fix to this issue 😉
If you want to give it a try, take a look at [this GitHub workflow](https://github.com/arduino/arduino-ide/pull/1547/checks): when it's finished, you'll be able to download a test build.

Thank you so much for reporting this issue 🤗

r3inbowari, 2022-10-07:
Thank you! It works great now.😊

closed 2022-10-21 by AlbyIanna