arduino/arduino-ide #1546: Serial port performs a reset to attached UNO on startup and shutdown of IDE 2.0.0

RalphBacon, 2022-10-07:


per1234, 2022-10-07:
Thanks for your report @RalphBacon.

As far as the reset on start up is concerned. It is expected that the board will restart on initialization of the Serial Monitor. Unlike Arduino IDE 1.x, Arduino IDE 2.x remembers whether you had Serial Monitor open in the IDE window of a given sketch and restores that state the next time you open the sketch, which will be done automatically if the sketch was open the last time you closed the IDE. So in this case the reset on startup is expected as the Serial Monitor initializes when the IDE window opens with Serial Monitor.

However, the reset on shut down is not expected to me. We had a similar report at https://github.com/arduino/arduino-ide/issues/1420, which unfortunately I have not been able to reproduce. In that case, the reporter associated the problem with the Arduino IDE Serial Monitor. I'm wondering whether this might also be a required condition in your case.

We can check that by seeing whether the reset still occurs even if Serial Monitor (and Serial Plotter while we're at it) was never used during the IDE session. Please try this and then comment here to report the results:

1. Start the Arduino IDE if it is not already running.
1. If multiple IDE windows are open, close all except one.
1. If the **Serial Monitor** tab is present in the panel at the bottom of the IDE, click the **X** button on its tab to close it.
1. Select **File > Quit** from the Arduino IDE menus.
1. Wait for the Arduino IDE start up to finish.
1. Select the board and port for the Uno from the Arduino IDE menus if it is not already selected.
1. Select **File > Quit** from the Arduino IDE menus.

Did the board reset on start up?

Did the board reset on shut down?

RalphBacon, 2022-10-07:
Thanks for the quick reply. 

I can confirm that turning the Serial Monitor **off** ensures the UNO board is not reset either at start up or shutdown. 😊

However, I do not support the assertation that "... it is _expected_ the board will restart on initialisation of the Serial Monitor". 😲

If the creators of the Arduino IDE prefer this behaviour, I would _expect_ this to be a user-selectable option to determine whether the initial state of the DTR/RTS line states should be on or off at start up.  I often run sketches for several hours (if not days) for testing. I would not expect that board to be arbitrarily reset the next time I happened to open the IDE (with the Serial Monitor open). I'm sure you can see the unfriendly behaviour here!

Perhaps this can be considered as an (urgent) feature suggestion?

per1234, 2022-10-07:
Thanks for your feedback. The problem of unexpected resets on IDE startup will be resolved as part of the planned Serial Monitor rework that moves it to a separate window (as in Arduino IDE 1.x and similar to Serial Plotter in both IDE series): https://github.com/arduino/arduino-ide/issues/289#issuecomment-1070964887

I would still be interested in investigating the reset on shut down, as this is not expected.

Please try this and then comment here to report the results:

1. Start the Arduino IDE if it is not already running.
1. Select the board and port for the Uno from the Arduino IDE menus if it is not already selected.
1. If the **Serial Monitor** tab is not already present in the panel at the bottom of the Arduino IDE, select **Tools > Serial Monitor** to open it.
1. Click the **X** button on the **Serial Monitor** tab at the bottom of the Arduino IDE window.
1. Select **File > Quit** from the Arduino IDE menus.

Did the board reset on shut down?

RalphBacon, 2022-10-08:
I followed your instructions exactly.

When I click the **X** on the Serial Monitor tab it closes. If I then close the IDE the UNO does NOT reset. It only resets if the Serial Monitor tab is open on shutdown.

I hope this helps you resolve the issue.

Good to hear that the overall behaviour of the Serial Monitor is being reassessed generally!

I shall pass on the caveat regarding the unexpected board resets to my viewers but, that notwithstanding, encourage them to download and use it in preference to any V1.x version. It is a huge step forward for C++ uses of all levels.