arduino/arduino-ide #1548: AppImage shows default icon on Debian 11 / Gnome 3.38.5

rei-vilo, 2022-10-08:
### Describe the problem

When launched, the `arduino-ide_2.0.0_Linux_64bit.AppImage` shows the default application icon on Debian 11 / Gnome 3.38.5.

### To reproduce

Launch `arduino-ide_2.0.0_Linux_64bit.AppImage`

### Expected behavior

Show the Arduino icon

### Arduino IDE version

2.0.0

### Operating system

Linux

### Operating system version

Debian 11 / Gnome 3.38.5

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-09:
Hi @rei-vilo. Thanks for taking the time to submit an issue.

Unless I misunderstand, I believe we have a previous report about this at https://github.com/arduino/arduino-ide/issues/131.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-10-09 by per1234

rei-vilo, 2022-10-09:
You are right! I did perform a search before but didn't found issue #131.

The issue is nothing critical, but useful.

Thank you!