arduino/arduino-ide #1549: Unable to upload to Arduino on M1 Mac Studio

ewrogers, 2022-10-08:
### Describe the problem

When using an Apple Mac Studio (M1 Max), I am unable to upload any code to the Arduino Nano via `/dev/cu.usbserial-AB0JQVPN`. Even the simple empty or blink script fails with the following error:

```
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
...
Failed uploading: uploading error: exit status 1
```

I have tried using different cables, Arduino Nanos, and even `avrdude` directly via CLI. The same error persists and the Nano does not seem to be responding.

Interestingly enough, using a ISP (with the same USB cable to it) allows me to upload and flash code/bootloader. I just cannot flash directly to the Nano. It is a genuine Nano and I have multiple that I have all tried, including flashing new bootloaders.

It seems to be related to the Apple Silicon and UART.

### To reproduce

1. Connect Arduino Nano to Mac Studio (M1 Max)
2. Select `/dev/cu.usbserial-AXXXXXXX` device
3. Select `Arduino Nano` as board
4. Attempt to upload blink example (or any other code)
5. Get error: `avrdude: stk500_recv(): programmer is not responding`
`avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00`

### Expected behavior

The code should be uploaded to the Arduino Nano successfully.

### Arduino IDE version

Version: 2.0.0 Date: 2022-09-14T07:08:28.861Z CLI Version: 0.27.1 [a900cfb2] 

### Operating system

macOS

### Operating system version

12.6

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-09:
Hi @ewrogers. Thanks for your report.

>using a ISP (with the same USB cable to it) allows me to upload and flash code/bootloader

Something you should be careful about is that when you do an "**Upload Using Programmer**" operation, the bootloader is erased. The error you are getting is expected when attempting a normal "**Upload**" operation to a board without a bootloader. So make sure to always do a "**Burn Bootloader**" operation before you switch back to attempting normal a "**Upload**" operation after using "**Upload Using Programmer**".

---

In order to gather more information that might help us to troubleshoot your problem, I'm going to ask you to provide the full output from the upload when in verbose mode.

Please do this:

1. Select **File > Preferences** from the Arduino IDE menus.
1. Uncheck the box next to **Show verbose output during: > `☑` compilation**.
1. Check the box next to **Show verbose output during > ☐ upload**.
1. Click the <kbd>**OK**</kbd>** button.
1. Select **Sketch > Upload** from the Arduino IDE menus.
1. Wait for upload to fail.
1. After the upload fails you'll see a button on the "**Upload error**" notification at the bottom right corner of the IDE window: <kbd>**COPY ERROR MESSAGES**</kbd>. Click that button.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here.
   In order to ensure the text is not corrupted, please use [code fencing](https://guides.github.com/features/mastering-markdown/).

ewrogers, 2022-10-09:
Interesting! I didn't know this about the bootloader being erased via ISP. It does seem like that has solved my issue with this. Great to know and now I feel like a bit of a dunce...

closed 2022-10-09 by ewrogers

per1234, 2022-10-09:
I'm glad it is working now. Thanks for taking the time to update the issue.

Regards, Per