arduino/arduino-ide #1552: Add cache clear option

Bodmer, 2022-10-09:


matthijskooijman, 2022-10-09:
I believe that `arduino-cli compile` has a `--clean` option that would achieve what you need.

However, I also think that the caching code should handle this automatically, e.g. by tracking the include paths used for compiling the core files and noticing that these change when compiling another sketch (this implies that sketches using `build_opt.h` or otherwise include sketch dir in the core include path can never share cached cores between different sketches, but that's just how it is, I guess).

Bodmer, 2022-10-09:
I tried setting:
```
cache.enable=false
compiler.cache_core=false
```
in preferences.tft but that did not work so perhaps this setting in now deprecated?

per1234, 2022-10-10:
Hi @Bodmer 

>in preferences.tft

Arduino IDE 2.x uses a completely separate preferences store than Arduino IDE 1.x. The `preferences.txt` file is not used at all by Arduino IDE 2.x. There is no setting in Arduino IDE 2.x to control caching behavior.

I saw you also posted about this on the Arduino Forum:

https://forum.arduino.cc/t/cache-clear-option/1040106

I'm looking forward to your opinion on the alternative proposal I described in my reply there.

However, I always prefer a solution that allows the caching to always be handled correctly automatically without any need for special action from the user, as **matthijskooijman** described.

I had transferred this issue from its original location in the `arduino/arduino-cli` repository to `arduino/arduino-ide` because the feature you request would only require changes in the Arduino IDE 2.x codebase (since Arduino CLI should already have the necessary capability). But if the solution is to be improving the caching behavior then the work would be done in the `arduino/arduino-cli` repository after all.

Bodmer, 2022-10-10:
Yes, an option would be great.

I have also got a corrupted cache in the past and restarting the IDE used to fix that.

At the moment my work around is to start the IDE with a batch file that clears the cache. But that is also not a clean way to do things. For example:
```
cd C:\Users\xxxxx\AppData\Local\Programs\Arduino IDE
"Arduino IDE" --clean
```

Yep, I was investigating CLI options and incorrectly pasted the issue there. Thanks for moving it to the correct place.

MCF, 2023-06-20:
In the various discussions around this feature found here and on various forum postings, there is an important and valid use case that is not mentioned - that is the case where compiler warnings are turned up or on for a sketch and then you wish to iteratively clean up the warnings.  If there are many warnings across many files it is unlikely that you will want to fix them all at once- better to fix them a few at a time.  In this case there is no easy way to force the recompilation of files that have successfully compiled - but with warnings. 

I understand the underlying reason that the Arduino IDE is meant to do everything it can automatically (for newer users among others).  But I frankly find the reluctance to have an option to do a complete clean of the cache of a sketche's files a bit extreme.  I get it - the less a new user has to worry about the better - but being able to rule in or out a caching problem would seem to be a good enough thing that it should be available somewhere in the IDE (hide it away in a dark corner, but please make it available :-) ).

FWIW my solution was to turn on the verbose option for compilation (in preferences) then figure out where the sketch's temporary directory was, and then just delete it in between compilations.  This is described in more detail in the forum link noted above.

Thanks to everyone who contributes code/effort to keeping the IDE going.  Apologies for my lack of any real help other than complaining.

MCF, 2023-06-21:
Coincidentally I ran across a related issue in the arduino-builder repo that discusses the use case I noted above.   For completeness, here is the issue:

https://github.com/arduino/arduino-builder/pull/301

thomascallahan, 2023-06-24:
I just had a similar problem due to renaming some .h and .cpp files in a project. Verify/Compile was still using the old, cached files. Took me a while to notice the old filenames in the error output, and then a bunch of googling trying to find a way to clear the cache (the switch boards, verify, switch back, worked for me).

I'm a relative newcomer to Aruduino, but have been a web developer for 25+ years so this kind of thing isn't exactly new to me. Once I saw the old filenames I immediately knew that there was caching of some kind. I can imagine a true newbie would be extremely confused by this. Having the option to clear the cache seems necessary, and/or having it happen automatically if you rename files in a sketch?

per1234, 2023-06-26:
> I just had a similar problem due to renaming some .h and .cpp files in a project. Verify/Compile was still using the old, cached files.

This is a known bug: https://github.com/arduino/arduino-ide/issues/970 https://github.com/arduino/arduino-cli/issues/1240

This is a case where the development resources would be much better devoted to fixing the bug instead of a adding a cache clear capability to allow the users to do something the build system should be doing automatically.